PiSupply/PiJuice #102: SYS_FUNC_REBOOT - just kills PiJuice.service

rohtua, 2018-04-25:
When setting a button to run the SYS_FUNC_REBOOT function all that seems to be happening is it kills the PiJuice.service. Clicking apply in the settings window displays the following error:
[image]

Checking the status of PiJuice.service shows:

`pijuice.service - PiJuice status service
   Loaded: loaded (/lib/systemd/system/pijuice.service; disabled; vendor preset:
   Active: inactive (dead)`

Then when I run `sudo service pijuice restart` it restarts the pijuice service and performs the reboot but until I restart the service nothing happens to turn the Pi on or off.. Is there a way to force the service to reboot after a crash? 

Also I've been trying to run user functions but they don't seem to work for me, I've tried typing the command I'm after direct into the user_func box and tried point to a .sh file but nothing happens when I press the button.

rohtua, 2018-04-28:
I've got it working, all I needed to do was run sudo systemctl enable pijuice that seems to have sorted the issue with SYS_FUNC_REBOOT

tvoverbeek, 2018-05-05:
Closing issue

closed 2018-05-05 by tvoverbeek