PiSupply/PiJuice #103: GUI shows "Communications Failed" after a few seconds

PeterOGB, 2018-04-26:
After ~30seconds of running the GUI shows "Communications Fail".
i2cdetect no longer shows the device on address 14.
I have to remove the battery before it will do any thing again.
Pi = 3B    Latest kernel and firmware (updated this morning). 

So far not a very good experience :-(

PeterOGB, 2018-04-27:
Very disappointed with whole thing. I had to remove the battery again this morning before it would do anything , then it stopped responding to I2C and button presses. Had to remove battery again to power down the PI.  PiJuice now removed from the PI.

rohtua, 2018-04-29:
I had similar issues, the problem for me was that the Pijuice.service kept crashing or wouldn't start on boot. For me it was crashing when the SYS_FUNC_REBOOT option was tried. If I ran `sudo service pijuice restart` then it would work and any button presses I'd made on the Pijuice would be completed I.e. Reboot. It may be worth checking the service status and restarting the service after pressing one of the buttons. If the service is 'dead' on startup running `sudo systemctl enable pijuice.service` seems to have done the trick for me. Maybe try restarting the service first once you've had the lost connection error and see if anything different happens from there. Hope this helps

tvoverbeek, 2018-05-05:
@PeterOGB Try the following:
In /boot/config.txt add the line `dtoverlay=i2c-bcm2708` and reboot.
This loads the older bcm2708 i2c driver instead of the upstream one (i2c-bcm2835).
It is known the bcm2835 driver has issues with some i2c peripherals due to bugs in the RPi i2c controller.
See also issue #108 where this change made PiJuice work.

closed 2019-09-15 by ChristopherRush