PiSupply/PiJuice #104: When I go to PiJuice Settings, it asks for raspberry pi password, and crashes

jleval2, 2018-04-26:
When I go to the "PiJuice Settings" under Preferences on Raspbian, the screen goes black, and then I am asked to login to my raspberry pi with the options "Shutdown" or "Login". After I login, I just go back to the main Raspbian screen, and nothing happens.

jleval2, 2018-04-26:
Does the same thing.

I should also mention, the battery level at the top right had disappeared. I tried uninstalling and reinstalling pijuice, which did not fix this problem, and then this problem began.

craigusus, 2018-04-27:
I received my board yesterday, installed software and connected this morning. Getting the same as @jleval2 , as soon as I click on Menu > Preferences > PiJuice Settings I am presented with what looks like a login prompt.
When I type in my password its as if it is a new login, as I have a script running in a terminal window on log on, that terminal window opens and runs my script.
Also with no icon in the taskbar.

dl8bb, 2018-04-27:
in the meantime I've uninstalled the pijuice packages 

pijuice-base/stable,now 1.3.2-1 all [installed]
pijuice-gui/stable,now 1.3.2-1 all [installed]

via 

$ sudo apt-get remove pijuice-base

and reinstalled the pijuice-base_1.3.1-1_all.deb and pijuice-gui_1.3.1-1_all.deb from the github repo via

$ sudo dpkg -i pijuice-base_1.3.1-1_all.deb pijuice-gui_1.3.1-1_all.deb

Now at least the pijuice-gui is working again. So there is some thing strange in version 1.3.2.

QL8-6754, 2018-04-27:
Getting the same this morning. This is what I have found so far. 

The PiJuice software was installed on a new, clean Raspbian (2018-04-18 Desktop). apt-get update was run before the PiJuice install but not apt-get upgrade. The PiJuice software could be started from the menu and the taskbar battery indicator appeared and occasionally sort of worked. However after apt-get update (the one where upon boot Raspbian Desktop tells you it has overwritten config files and stored them in oldconfig) I got the issue where there is no taskbar indicator and using the menu sent me to a login screen. Uninstalling the PiJuice software and reinstalling did not help.

dl8bb, 2018-04-27:
Can confirme after **downgrade to version 1.3.1** all went to normal. So my advice would be remove the latest installed/updated pijuice packages (version 1.3.2 in my case), download **pijuice-gui_1.3.1-1_all.deb** and **pijuice-base_1.3.1-1_all.deb** from https://github.com/PiSupply/PiJuice/tree/master/Software/Install and install via **dpkg -i** as mentioned above.

hoopsurfer, 2018-05-03:
+1 on password request trying PiJuice Settings after install on latest 4-18 Raspian (fully apt updated/full-upgraded then install PiJuice per github instructions). The panel item added was for a pi-top battery not PiJuice and it shows 0%.  Will downgrade.

jleval2, 2018-05-03:
Installing 'sudo apt-get install python-gtk2' "half" worked. The crash no longer happens when I go to setting, but the tray icon is corrupt. It shows only when the PiJuice is plugged in (after 30-40 second). It then disappears when the PiJuice is unplugged from a power source.

tvoverbeek, 2018-05-03:
See my last sentence above and Issue #108

craigusus, 2018-05-04:
I have just installed the Rasbian 2018-04-18 image fresh from [raspberrypi.org](https://www.raspberrypi.org/downloads/raspbian/)
I then ran:
```
sudo apt update
sudo apt full-upgrade
```
rebooted
```
sudo apt install python-gtk2
```
to which I was informed that it was already up to date
```
sudo apt autoremove
```
as I was promoted to do so
````
sudo apt install pijuice-gui
````
Once completed I went to Menu > Preferences > PiJuice Settings and it logged me out again. But once I logged back in the icon was in the task bar that I could open the settings with. Plus when going back to Menu > Preferences > PiJuice Settings it opens and works, seemingly, fine. Even after 3 reboots all boots up working.

Am guessing that straight after installing pijuice-gui and rebooting all will work fine.

tvoverbeek, 2018-05-04:
If after the initial apt-get update/upgrade you do `sudo apt autoremove` the python-gtk2 package gets removed causing the log out/log in problem when starting pijuice_gui.
@craigusus your `sudo apt install python-gtk2` set the package to manually installed.
Therefore it was not removed by the following `sudo apt autoremove`.

craigusus, 2018-05-18:
Can the [installation instructions](https://github.com/PiSupply/PiJuice#installing-the-software) be updated to make sure `python-gtk2` is marked as manually installed/not removed?
i.e: `sudo apt install python-gtk2 pijuice-gui`

tvoverbeek, 2018-05-18:
In the 1.3.3 debian packages (now available from the raspbian archive) python-gtk2 is added as a dependency for pijuice-gui. Should take care of the problem.

tvoverbeek, 2018-06-03:
Closing the issue, since 1.3.3 is available for some time.

closed 2018-06-03 by tvoverbeek