PiSupply/PiJuice #105: PiJuice bricked after V1.2 firmware upgrade

wboerhout, 2018-04-27:
PiJuice connected to a Raspberry Pi Model B+ V1.2 running Raspbian Stretch Lite (latest update) and pijuice-base.
Worked fine until I downloaded and installed V1.2 firmware with 

`pijuiceboot 14 PiJuice-V1.2_2018_03_28.elf.binary`

The upgrade process went fine, but now the pijuice board is completely unresponsive, no lights, no power to the Pi. All communication with python scripts result in "COMMUNICATIONS ERROR"

How do I fix this?

EricS61, 2018-05-06:
Having an issue with firmware update/upgrade as well. Pi/PiJuice is shut down. LED near SW3 is steady blue, and LED near SW1 flashing between Green & Blue. Instructions say to press SW3 before pressing SW1. I've tried press and hold SW3 while momentarily pressing SW1, momentarily pressing both SW3 & SW1, and press and hold both SW3 and SW1 with the Pi not powering up for any combination.

tvoverbeek, 2018-05-06:
Which instruction?
A short press on SW1 alone (the one closest to the green & blue flashing led) should power the Pi.
The long press on SW3 is to force the start of the bootloader, to be used with the pijuiceboot utility.
From the README on the hardware page (https://github.com/PiSupply/PiJuice/tree/master/Hardware):
```
**Special functions**

- Dual long press of SW1 and SW2 for 20 seconds will reset PiJuice HAT configuration to default.
  This applies to the MCU configuration only.
- Holding pressed SW3 while powering up PiJuice will initiate the bootloader. This is used only
  in cases when ordinary initiation through I2C does not work because of damaged firmware.
```

EricS61, 2018-05-06:
OK, had to press and hold SW1 until the power LED on the Pi extinguished. Pressed and held SW3 while momentarily pressing SW1 powered up the Pi. Now, LED next to SW3 is still solid blue. How do I rectify this? In the GUI, if I select Configure HAT, and select Firmware tab, there's a notification that New firmware (V1.2) is available. I've clicked on the Update firmware button perviously, clicked on the "OK" for the warning, and the completion. Do I just reboot?

EricS61, 2018-05-06:
Rebooted Pi after firmware update. SW3 LED is now solid blue in color. PiJuice settings > HAT tab indicate a Fault "battery profile invalid charging temperatureCOOL". Selecting Configure HAT tab, the PiJuice HAT Configuration pop-up only shows General, Buttons, & LEDs tabs.

wboerhout, 2018-05-06:
The jury is still out on the resolution of my original question. My PiJuice does not communicate anymore after updating it with the V1.2 firmware.

At one point, I saw output from  

`i2cdetect -y 1`

as follows:
```
    0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f

00:          -- -- -- -- -- -- -- -- -- -- -- -- --

10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --

20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --

30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --

40: -- 41 -- -- -- -- -- -- -- -- -- -- -- -- -- --

50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --

60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --

70: -- -- -- -- -- -- -- --
```
but the 41 (instead of the expected 14) disappears after the first comm attempt. I still go for 'bricked'.

wboerhout, 2018-05-06:
```
pi@raspi1 ~ $ cat /boot/config.txt
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# https://github.com/PiSupply/PiJuice/issues/105
dtoverlay=i2c-bcm2708
pi@raspi1 ~ $ i2cdetect -y 1

     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

EricS61, 2018-05-06:
I gave up on the latest firmware update (it was draining the PiJuice battery even with the Pi powered down), and have backed down to firmware that was uploaded/installed (V1.1_2018_01_15) when I installed PiJuice application. Battery is now charging, I have all the tabs again in PiJuice HAT Configuration, and no faults.

wboerhout, 2018-05-06:
I have all but given up on the PiJuice. Reverting to old firmware would be my next choice, but since the PiJuice is incommunicado since loading V1.2, nothing at all can be loaded now.

EricS61, 2018-05-07:
wboerhout, not sure if this will help, but this is what I did;
Press and hold SW1 till the red power light on the Pi turns off.
Ensure the power is plugged into the PiJuice, not the Pi itself.
Quickly press and release SW1. Pi will power up.
Either ssh into the Pi, or open up a terminal window
cd /usr/bin
./pijuiceboot 14 /usr/share/pijuice/data/firmware/PiJuice-V1.1_21018_01_14.elf.binary
sudo reboot
Bring up the PiJuice app. You will have to adjust some settings to suit your tastes/needs.

wboerhout, 2018-05-07:
Last week, I had a lengthy conversation outside this forum with Chris from Pi Supply. All this came and went. For good measure, I retried it now the alternative i2c driver is loaded, but no deal:

```
pi@raspi1 ~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- 41 -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```
Note the weird "41" where 14 is expected. I guess it is random data. Then:

```
pi@raspi1 ~ $  pijuiceboot 41 /usr/share/pijuice/data/firmware/PiJuice-V1.1_2018_01_15.elf.binary
using address 41
Input file /usr/share/pijuice/data/firmware/PiJuice-V1.1_2018_01_15.elf.binary
Starting bootloader
Starting bootloader old firmware
Sending get command
error receiving data -1
```
No communication. Then, after that:
```
pi@raspi1 ~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```
Same thing happened when I used "14" as address regardless of the 41 that i2cdetect reported. For all that matters, the PiJuice just isn't there. And, as stated in my original post, I have not seen one LED light up on the PiJuice after flashing V1.2.

wboerhout, 2018-05-07:
pijuice-base (1.3.2-1) has replaced 1.3.1-1 after apt-get upgrade. No changes in behaviour after this.

tvoverbeek, 2018-05-07:
Managed to get my PiJuice in a state where it reports only 0x41 and also with no replies at all in i2cdetect.

Try this to recover:
- PiJuice mounted on the RPi, no power applied.
- Remove the BP7X battery
- While keeping SW3 (the middle one of the three switches) pressed apply power to the RPi (not the PiJuice)
  LEDs on the PiJuice should remain off when the RPi boots.
- The PiJuice is now in bootloader mode
- Run 'pijuiceboot 14 <your favorite firmware>' in a terminal or via ssh (No initial i2cdetect to test)

For me pijuiceboot runs normally and then starts the just uploaded PiJuice firmware.
In case you used the 1.2 firmware you should get the blue LED2 and a red or red/blue LED1 (since no battery connected)
For the 1.1 firmware you do not get the blue LED2.

Hope this helps.

wboerhout, 2018-05-07:
Brilliant, thanks! I would never have come up with this sequence.

One thing though. I used this procedure to load the V1.1 firmware, and checked that it was working OK. After that, I repeated it with a load of the V1.2 firmware. It appears to flash prpewrly with pijuiceboot, but after ebooting and inserting the battery, it still reports V1.1-0 as firmware version.

LEDs and function appear to be OK. 

I have reverted to the original confix.txt, without `dtoverlay=i2c-bcm2708`

Testing with pij.py:
```
#!/usr/bin/python
# -*- coding: UTF-8 -*-
from __future__ import print_function

import time
from pijuice import PiJuice

# ---
pijuice = PiJuice(1,0x14)

print(time.strftime( "%Y%m%d%H%M%S", time.localtime( time.time() ) ) )

firmware = pijuice.config.GetFirmwareVersion()
if firmware['error'] == 'NO_ERROR':
   fvers = firmware['data']
   print('firmware: V', fvers['version'], '-', fvers['variant'])
else:
   print('firmware: ', firmware['error'])

charge = pijuice.status.GetChargeLevel()
if charge['error'] == 'NO_ERROR':
   charge = charge['data']
   print('  charge: ', charge, '%')
else:
   print('  charge: ', charge['error'])

temp =  pijuice.status.GetBatteryTemperature()
if temp['error'] == 'NO_ERROR':
   temp = temp['data']
   print('    temp: ', temp, '°C')
else:
   print('    temp: ', temp['error'])

vio =  pijuice.status.GetIoVoltage()
if vio['error'] == 'NO_ERROR':
   vio = vio['data']
   print('  iovolt: ', float(vio)/1000.0, 'V')
else:
   print('     vio: ', vio['error'])
```
with output: 
```
pi@raspi1 ~ $ python pij.py
20180507204850
firmware: V 1.1 - 0
  charge:  63 %
    temp:  37 °C
  iovolt:  5.038 V
```

tvoverbeek, 2018-05-07:
The V1.2 firmware reporting itself as V1.1 is a known problem (See Issue #95).
Looks it is working again for you.
Are you only using pijuice-base?
If so, the pijuice_cli.py now on GitHub has a lot of bug fixes including proper support of new firmwares.
By the way, assuming your are Dutch like me.
Wens je het allerbeste met je PiJuice.

wboerhout, 2018-05-07:
My Pi for this project (mounting a dashcam on my robot lawn mower) doesn't need a GUI so yes, I will look into that, and into pijuice_cli.py as well.

Nogmaals, hartelijk dank.

shawaj, 2018-05-14:
@wboerhout we have a new firmware out now which fixes the reporting wrongly issue.

we should also have a package update in the coming week which will fix all the issues and pull the new firmware in also

thanks for your patience and thanks @tvoverbeek for helping!!

rbnpi, 2018-05-17:
Just loaded the new V1.2_2018_05_02.elf.binary using 
pijuice 14 V1.2_2018_05_02.elf.binary
Says it went successfully but my board now is totally unresponsive.
(I also loaded the new 1.3.3 deb which seemed to work OK.
Where do I go from here?

shawaj, 2018-05-17:
@rbnpi could you start this as a new issue please? Did you try updating the firmware from inside the GUI first? The manual update is not recommended unless someone tells you to do that

shawaj, 2018-05-17:
Also did you try the steps above posted by ton?

wboerhout, 2018-05-17:
Well, in all fairness, there abosolotely should be a documented way to update without a GUI. None of my projects need a GUI. I have never installed a GUI environment on a Raspberry and I am not planning to do that.

tvoverbeek, 2018-05-17:
The pijuice_cli.py in the pijuice_base 1.3.3 package does a proper update.
No gui needed.
Until the 1.3.3 packages are available in the raspbian archive you will
have to download the .deb from GitHub.

rbnpi, 2018-05-17:
Well I updated because received a general email sent today which suggested it. I put the 1.3.3 deb in but couldn't upgrade with that, so reverted to using pijuice  to do it. This said it installed OK. However now no response from battery leds although they were doing weird things for a bit, and no response from i2cdetect -y 1.

shawaj, 2018-05-17:
Have you read the way this was resolved by Ton above?

rbnpi, 2018-05-17:
OK I used a previous SD card and managed to restore the old firmware using Ton's method. I'll leave it there until the new apt-get packages are released!

shawaj, 2018-05-17:
Glad to hear it's working again

wboerhout, 2018-05-18:
For my part, I downloaded and installed pijuice_base 1.3.3, then using pijuice_cli.py I upgraded the firmware to V1.2 . All went well, I'm happy

shawaj, 2018-05-18:
@wboerhout awesome! glad to hear it. pesky electronics seem to be a bit tempremental. Enjoy your PiJuice and your weekend!

tvoverbeek, 2018-07-11:
It seems you are using the wrong firmware file. The correct one (as in the current 1.3.3 pijuice-base package) is `PiJuice-V1.2_2018_05_02.elf.binary`
Also this version programs 276 pages instead of 157.
Here is my output:
```
pijuiceboot 14 ./PiJuice-V1.2_2018_05_02.elf.binary 
using address 14
Input file ./PiJuice-V1.2_2018_05_02.elf.binary
Starting bootloader 
Starting bootloader old firmware
Sending get command 
bytesRead 13  dataLen 11
ack 79
data: b, 10, 92
data received 11
bootloader version: b
commands: 10, 0, 1, 2, 11, 21, 31, 44, 63, 73
erase page count 35
first page erase succcess 0
erase success 0
page count 276
mem data read 256, 256
Page 275 programmed successfully
mem data read 256, 256
Page 274 programmed successfully
mem data read 256, 256
......
Page 4 programmed successfully
mem data read 256, 256
Page 3 programmed successfully
mem data read 256, 256
Page 2 programmed successfully
mem data read 256, 256
Page 1 programmed successfully
mem data read 256, 256
Page 0 programmed successfully
EEPROM programming finished successfully
Code executed successfully
``` 
Suggest you try to reflash using the correct firmware (get it from github or the 1.3.3 pijuice-base package).
When the flashing is successfull, the firmware restarts and LED D2 should light bright blue.

tvoverbeek, 2018-07-11:
> Previously I did the download with curl
>
> curl
> https://github.com/PiSupply/PiJuice/blob/master/Firmware/PiJuice-V1.2_2018_03_28.elf.binary
> -o PiJuice-V1.2_2018_05_02.elf.binary
>
> and the file size was 40067, with git clone the files size is 70452.
> Can it be a good idea to put a zip version in the repository ?
> Or do you have another way for doing it ?

Use the git clone one. 70452 is the correct size.
Note with curl you downloaded the March 28 version, while the May 2nd
version is the correct one.

tvoverbeek, 2018-07-11:
Seems your curl download truncated the file. The correct size for the March 28 version is 70496, but please use the May 2nd version.

closed 2018-10-19 by wboerhout