PiSupply/PiJuice #106: COMMUNICATION_ERROR

EricS61, 2018-04-29:
After almost a 3 year wait, the PiJuice arrived via post. Connect it up to my Pi 2 model B. Installed the pijuice-gui as per instructions (pijuice-base is also installed as part of the package). Rebooted the Pi. The PiJuice icon in the panel is not visible, but if I hover over the space it's occupying and right mouse click, I get 'Settings' and 'About'. About says Software version: 1.3.2. If I click on Settings, every setting says "COMMUNICATION_ERROR". Removing the battery and re-seating it, does nothing. Going through the various tabs within settings results in a pop-up saying "COMMUNICATION_ERROR". Running Raspbian Jessie Stretch with Linux kernel 4.14.34-v7+. I did a fresh install on a new SD card. Ran update, upgrade, and rpi-update.

tvoverbeek, 2018-04-30:
For a fix of the pijuice_tray problem see my comment in issue #108

tvoverbeek, 2018-05-05:
Closing this one. Continued in #108

closed 2018-05-05 by tvoverbeek