PiSupply/PiJuice #107: Questions arising from time/charge graph

jdotbdot, 2018-04-29:
This graph (blue line "Power") shows the charge level reported by "pijuice.status.GetChargeLevel()" against time.
This is a Pi 3B+ running headless and idle apart from reporting these stats every 10 seconds.
It shows that the Pi stayed alive on battery power for almost 3 hours, from 19:00 to 21:51, pretty good performance I think.
Questions relating to the red annotations are below.
[image]

1. USB power pulled at 19:00. I suppose battery voltage (and so the reported charge level) is bound to drop when the load is applied?
2. What is the cause of these big short term drops in reported charge level? There is no matching pattern in the CPU idle percentage.
3. The Pi actually managed to report a charge level of 0 before shutting down. Is it bad for the battery to discharge it this far? (Pijuice configuration is the default.) If so, what minimum charge level do people recommend? How can I set this using command line tools on a headless Pi?
4. USB power reapplied at 22:00 - plugged in to the PiJuice, not the Pi. The Pi booted immediately. I wonder if it's better for the default to be wait until the battery has some charge?
5. The initial reported charge rate seems improbably rapid. 
6. Usb power pulled as soon as reported charge lvel hit 100. The immediate drop from 100 to 70 suggests that the reported charge level is exaggerated by 25 - 30 percentage points when it is charging. 
Is it possible to tweek the software to address 5 & 6?

ChristopherRush, 2018-04-30:
1. Yes, as load is applied it then calculates the battery levels. See header "Battery charge/discharge" [https://github.com/PiSupply/PiJuice/tree/master/Hardware](url)

2. Honestly, I don't know. It could be interesting to monitor the GPIO power input to see if there is a correlation.

3. No, perfectly fine to run it down to 0% but you are risking the Pi not shutting down properly if the current is high. I would recommend a minimum level of 3-5%.

4. You can configure this in the settings by setting a "wake up on charge %", usually this would be a higher percentage value. 

5. Charge level percentage is not calculated with current load in mind, unlike the discharging process. It is solely based on actual voltage levels. So charge levels would most likely see a spike from the start. 

6. As per the previous answers, charging/discharging percentage values are calculated differently.

erichbieri, 2018-07-05:
I did a similar test like jdotbdot did with a 1200mAh single cell Lipo battery attached to the Pijuice, but I attached a MAX17043 FuelGauge Chip in parrallel to the battery. Here are my findings:

1. As soon as I plug the USB cable the state of charge reported by the pijuice.status.GetChargeLevel() method drops from 100% to about 80%. Similar to point 1) stated by jdotbdot. The state of charge reported by the MAX17043 gradually decrease (no initial drop).
2. I discharged the battery down to 3.4 volts till the MAX17043 returned a state of charge of 0%. The  pijuice.status.GetChargeLevel() returned 0% long before the battery was a 0%! 
3. When I reapplied the usb charger. The state of charge returned by the pijuice.status.GetChargeLevel() jumped to approx. 70%. Similar to point number 5) stated by jdotbdot. The state of charge reported by the MAX17043 here again gradually increased (no sudden increase of the %). 

My Questions: 
1. Does the pijuice.status.GetChargeLevel() method directly reflect the the result of the RSOC (0x0D) register of the LC709203F or is there some intermediate calculation performed on top of this?
2. Is there a way to directly connect to the I2C bus of the PiJuice?
3. Is there a way to improve this behavior?

ChristopherRush, 2019-09-15:
Introduced alternative to fuel gauge ic way of remaining state of charge estimation, together with fixes of fuel gauge estimation in Firmware v1.3 - f571b4384aea02c75073204db5213538a9c02fbf

closed 2019-09-15 by ChristopherRush