PiSupply/PiJuice #108: Missing battery level indicator

woodentop, 2018-04-30:
Running pijuice 1.3.2.  When on battery power pijuice_tray.py is searching for icons which are not present in /usr/share/pijuice/data/images. For instance, running pijuice_tray.py from the command line: 
```
84%
/usr/share/pijuice/data/images/bat-84.0.png
icon file not exist
```
This is wrong for two reasons: there is no level 84 (only 80 and 90) and the appended ".0" is not present in these files:
```
/usr/share/pijuice/data/images $ ls
bat-0.png    bat-90.png      bat-in-80.png    bat-rpi-70.png
bat-100.png  bat-in-0.png    bat-in-90.png    bat-rpi-80.png
bat-10.png   bat-in-100.png  bat-rpi-0.png    bat-rpi-90.png
bat-20.png   bat-in-10.png   bat-rpi-100.png  connection-error.png
bat-30.png   bat-in-20.png   bat-rpi-10.png   no-bat-in-0.png
bat-40.png   bat-in-30.png   bat-rpi-20.png   no-bat-rpi-0.png
bat-50.png   bat-in-40.png   bat-rpi-30.png   pijuice.png
bat-60.png   bat-in-50.png   bat-rpi-40.png
bat-70.png   bat-in-60.png   bat-rpi-50.png
bat-80.png   bat-in-70.png   bat-rpi-60.png
```

This shows up in two circumstances:
1.    Boot on battery - no icon on panel but you can still mouseover to get %ages
2.    Pull power - you are left with whatever icon last displayed - which does not change. This is, of course, misleading, although mouseover still works.

I'm using 1.3.2. I don't know if this occurs in 1.3.1. If 1.3.1 is OK then 1.3.2 may have changed the icon naming scheme but forgot to package icons with the new names?

tvoverbeek, 2018-04-30:
Problem with pijuice_tray is due to a missing // for python3 compatibility. It is already fixed in GitHub. Replace /use/bin/pijuice_tray.py with the current GitHub version for the time being. With the GitHub version the tray icon also works when on battery power and with power connected to the RPi.

woodentop, 2018-04-30:
Ah yes - so you get only the tens - I see :)

EricS61, 2018-04-30:
Uninstalled both pijuice-gui and pijuice-base, rebooted, and reinstalled (sudo apt-get install pijuice-gui). Looks like the fix isn't making it to the install package. For curiosity sake, I browsed the /usr/bin/pijuice_tray.py file with the pijuice_tray.py file located in the Software/Source folder and couldn't find the missing // .

tvoverbeek, 2018-05-01:
Sorry, not yet on the main branch. See  my PR #94.

EricS61, 2018-05-02:
Shut down the Pi completely yesterday to put it into a PiJuice case. Connected everything back up, powered it up, started up my favorite app that I like to run. Check on the Pi to see how all is running, and low and behold the battery icon is now showing up in the panel. Hover the mouse pointer over it, and it says 97%. Now to figure out the actual PiJuice app. Still showing COMMUNICATION_ERROR even when I play with any of the configuration settings.

EricS61, 2018-05-05:
Installed the latest (2018-04-18) on a new SD card. Ran update/upgrade/install pijuice-gui. Edited the pijuice_tray.py and placed the extra / in lines 157 & 159 as mentioned in another posting. Rebooted. Icon is still missing (invisible). Run the gui and COMMUNICATION_ERROR still showing up. So, I'm at a standstill. Do I remove 1.3.2 and attempt to install 1.3.1 which others have mentioned success, or hope that one day this will get fixed. So, rather than pull more hair out. I'm stopping for the night, getting another cold liquid refreshment, and may attempt another crack at repair, or start from scratch with another SD card.

tvoverbeek, 2018-05-05:
@EricS61 First try the folowing:
In a terminal window start python and then
```
pi@raspberrypi:~ $ python
Python 2.7.13 (default, Nov 24 2017, 17:33:09) 
[GCC 6.3.0 20170516] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> from pijuice import PiJuice
>>> pijuice=PiJuice(1,0x14)
>>> 
```
In your case I expect the `pijuice=PiJuice(1,0x14)` statement to give an error.
The error message might give a clue about the problem

Also (if you have i2c-tools installed) try `i2cdetect -y 1`.
The normal output looks like this:
```
pi@raspberrypi:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- 14 -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
pi@raspberrypi:~ $ 
```
0x14 is the PiJuice I2C address. The 'UU' at 0x68 is the simulated real-time-clock in the
PiJuice firmware. 'UU' because the RTC is claimed by the kernel.

Last question: Are you running as user pi or as an other user?

EricS61, 2018-05-05:
Did the terminal window entires. i2c-detect -y 1 responds with 14 at the 0x14 address, and 68 (not UU) at the 0x68 address. I'm running as user pi.

tvoverbeek, 2018-05-05:
What about the interactive python (the pijuice=PiJuice(1,0x14) thing)?
Are you on Jessie?

EricS61, 2018-05-05:
Yes on Jessie (2018-04-18). Fresh install on a brand new SD card. Did the interactive python entries, and no error(s). Not sure if that's a good thing, or not! lol

tvoverbeek, 2018-05-05:
OK. Seems some communication is working.
Next thing to look at is if the pijuice.service is working.
What is the output from `systemctl status pijuice.service`?
Mine (on Stretch) looks as follows:
```
pi@raspberrypi:~ $ systemctl status pijuice.service
● pijuice.service - PiJuice status service
   Loaded: loaded (/lib/systemd/system/pijuice.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2018-05-04 19:51:55 UTC; 17h ago
 Main PID: 356 (pijuice_sys.py)
   CGroup: /system.slice/pijuice.service
           └─356 /usr/bin/python /usr/bin/pijuice_sys.py

May 04 19:51:55 raspberrypi systemd[1]: Started PiJuice status service.
pi@raspberrypi:~ $ 
```
If the service is not running certain things do not work in the GUI.

tvoverbeek, 2018-05-05:
Next thing is to start the GUI manually in a desktop terminal window:
`python /usr/bin/pijuice_gui.py`.
Hopefully some error messages show up to pinpoint where the communication problem is.

EricS61, 2018-05-05:
Gui popped up. Battery - COMMUNICATION_ERROR, GPIO power input - COMMUNICATION_ERROR, USB Micro power input: nothing, Fault: COMMUNICATION_ERROR, System switch: off.

tvoverbeek, 2018-05-05:
Anything in the terminal window?

EricS61, 2018-05-05:
Nothing, nada, zilch.

tvoverbeek, 2018-05-05:
Same communication error when you run pijuice_cli.py in a terminal window (or via ssh)?
Just select Status after starting pijuice_cli.py

EricS61, 2018-05-05:
Getting communication error directly (keyboard, mouse, monitor), and remote into it via VNC. I just ssh'd into it, and what looks like an old ms-dos display. When I select "Status", it terminates with over a dozen Traceback entries. If I run it again, and select "General", I get a "Unable to connect to device" < OK  >. Pressing Cntl-C takes me back to prompt. I can keep running through each main selection if you like to see what replies I get for each if it will help?

tvoverbeek, 2018-05-05:
Can you somehow capture the Traceback entries? That might give me a clue what is happening.
Otherwise try the following in a terminal window (or via ssh):
```
pi@raspberrypi:~ $ python
Python 2.7.13 (default, Nov 24 2017, 17:33:09) 
[GCC 6.3.0 20170516] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> from pijuice import PiJuice
>>> pj=PiJuice(1,0x14)
>>> pj.status.GetStatus()
{'data': {'battery': 'NORMAL', 'powerInput5vIo': 'NOT_PRESENT', 'isFault': True, 'isButton': False, 'powerInput': 'NOT_PRESENT'}, 'error': 'NO_ERROR'}
>>> 
```
The pj.status.GetStatus() is the low level call to get the status on the first GUI screen.
Also here I would expect you get an error with Traceback messages

EricS61, 2018-05-05:
OK, have a screenshot of the pijuice_cli.py Tracebacks. I also have a screenshot of error when I type pyton in the terminal window (via ssh). How do I get the images to you? Don't see a way of attaching them to this.

EricS61, 2018-05-05:
And yes, I'm getting an error when I run python in a terminal window via SSH, but not when I'm connected to the pi directly, or remotely via VNC.

EricS61, 2018-05-05:
Just found your email address in your profile. Would you like me to email it to that address, or do you prefer another method?

tvoverbeek, 2018-05-05:
Just drag the files into the comment window (when you open the issue on github (https://github.com/PiSupply/PiJuice/issues/108)

EricS61, 2018-05-05:
[image]
[image]

EricS61, 2018-05-05:
This is what I get when I ssh into the pi and run the pijuice_cli.py

[image]

EricS61, 2018-05-05:
Sorry about that.
[image]

tvoverbeek, 2018-05-05:
OK, This is what I expected. You get COMMUNCIATION ERROR imediately.
By the way you must be on Stretch, not Jessie, since there is no 2018-04-18 version of Jessie.

Can you try the following:
In /boot/config.txt add the following line:
`dtoverlay=i2c-bcm2708` and reboot
This loads the previous driver for the i2c bus.
I have seen other projects where this was needed to get i2c working again.
You are on an older, slower Pi (2B). For me it works both on a 3B and 3B+.
After the reboot first try the short interactive python session again.

EricS61, 2018-05-05:
Yes I'm on Stretch. I've seen elsewhere reference to Jessie (Stretch), and continued the "reference" without reading up and seeing that they are two separate versions. My bad on that one! lol

So, made the change to the /boot/config.txt file, saved it, rebooted, and did another ssh. Attached is what I'm seeing now.
[image]

EricS61, 2018-05-05:
Sorry, attached the wrong image. I edited, removed the old one, and added the new one.

tvoverbeek, 2018-05-05:
The image I see is identical to the old one. Correct?
I am still seeing COMMUNICATION ERROR.

Just to check if the config.txt change worked, check the output from `lsmod` in a terminal.
It should have a line with  i2c_bcm2708 and not i2c_bcm2835.

How do you power it, via the RPi or the PiJuice USB?
Do you have any LEDs on the PiJuice (It should flash blue when charging)?

EricS61, 2018-05-05:
Yes on still seeing COMMUNICATION_ERROR. No, the config.txt change didn't work (see image below);
[image]

I power it through the PiJuice USB, yes I have a flashing blue/green LED.

Think I found my problem - dtoverlay=l2c-bcm2708 vice dtoverlay=i2c-bcm2708 in the config.txt file. Just edited file, and rebooting.

EricS61, 2018-05-05:
SUCCESS!!!!!!! ran interactive python via SSH, and returns were same as yours (no COMMUNICATION_ERROR). So, fired up VNC and the gui, and am now getting all kinds of information within the PiJuice Settings display, and the PiJuice icon is now displaying again up in the tray! :) :)

EricS61, 2018-05-05:
So, looks like the dtoverlay=i2c-bcm2708 entry in the config.txt. Funny, when I get ready to add another hat (amateur radio modem), I would have probably made that entry in the config.txt file as it also uses the i2c. So, more notes to scribble down for future reference. At least for me, I guess we can close out these issues (at least for me). Thanks again Ton!

tvoverbeek, 2018-05-05:
Glad we could find a solution.
Will also check on a Pi 2B which I have lying around.
Will probably result in a recommendation to use i2c-bcm2708.

Good luck with your ham project and closing the issue.

closed 2018-05-05 by tvoverbeek

KeithSloan, 2018-05-05:
I also have the missing/non displayed icon, but when I hover over bar I get a value.
Tried /usr/bin/pijuice_tray.py from github and the i2c-bcm2708 change ( Not getting comms errors as far as I know ). Neither worked.

Dave-Kay, 2018-08-23:
I have maybe a similar problem, in pijuice-base i get a "unable to connect to device" but different from the problem shown here, there is no led-activity whatsoever.