PiSupply/PiJuice #109: Installed pijuice-gui but no options on desktop.

KeithSloan, 2018-05-01:
Have installed pijuice-gui but do not see any icons or related options under preferences.

Tried a reboot but nothing.

cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

ChristopherRush, 2018-05-01:
Looks like you using an old OS version. Please update to the latest OS Raspbian Stretch as per the Raspberry Pi website [https://www.raspberrypi.org/downloads/raspbian/](url)

KeithSloan, 2018-05-01:
I have a number of things setup on my headless Pi and checking on the web upgrading to Stretch looks like a lot of hassle that I don't need at the moment. I just need the pijuice to act as a UPS. Will the hardware do this without the software?

tvoverbeek, 2018-05-01:
Should work with Jessie. Now on travel. Will check when back home.

tvoverbeek, 2018-05-01:
You do not need the GUI for an UPS setup. You do need pijuice.py from the pijuice-base package. See this project http://frederickvandenbosch.be/?p=2876 for an example.

tvoverbeek, 2018-05-04:
@KeithSloan For documentation the best place for now is github (https://github.com/PiSupply/PiJuice) especially the READMEs in the Hardware and Software directories.
The quick start guide is the one which was in the package. You can find it here: https://github.com/PiSupply/PiJuice/blob/master/Documentation/PiJuice%20Guide.pdf

Since you say you have a battery icon now with zero charge you seem to have it working partly.
Zero charge: Have you removed the plastic between the battery and the terminal block?

I did an install of Jessie (2017-07-05 full version) and ran `sudo apt-get update/upgrade` followed by
`sudo apt-get install pijuice-gui`.
For me the gui works under Jessie

KeithSloan, 2018-05-05:
Problem resolved now that I could find the info on the website. Thanks

tvoverbeek, 2018-05-05:
@KeithSloan OK, closing the issue and good luck with your PiJuice

closed 2018-05-05 by tvoverbeek