PiSupply/PiJuice #110: Error on running gui - "Failed to use I2C Bus"

notGMman, 2018-05-01:
I am receiving a popup window "Failed to use I2C bus. Check if I2C is enabled" when running the gui. I2C is enabled from raspi_config.

I'm running the very latest release of stretch in a Pi 3 (not plus) - apt-get update and upgrade performed before installing the pijuice software. Done an uninstall and purge and a reinstall - still the same message.

The device has been powered for a couple of days now. The LED is green, occasionally flashing blue. There are no errors when installing the pijuice software or in the raspbian logs. One process is running for pijuice - pijuice_sys.py and that process happily restarts via systemctl.

Running "gpio readall" shows the pins linked to I2C. "i2cdetect" is showing the two Pijuice devices. 

Any help greatly appreciated!

notGMman, 2018-05-03:
Hi Ton
Thanks for the advice, but it didn't work. I left the battery out and the pijuice inmounted for 20 mins and see the same behaviour when it's reassembled and powered up. 
Is there anything else worth trying? 
Thanks, David

tvoverbeek, 2018-05-03:
The 'Failed to use I2C bus. Check if I2C is enabled' message happen very early in pijuice_gui.
Can you try the following in a terminal window (or ssh session):
```
pi@raspberrypi:~ $ python
Python 2.7.13 (default, Nov 24 2017, 17:33:09) 
[GCC 6.3.0 20170516] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> from pijuice import PiJuice
>>> pijuice=PiJuice(1, 0x14)
>>> 
```
In your case the `pijuice=PiJuice(1, 0x14)` statement should throw an error.
That might help us further.

notGMman, 2018-05-04:
Hi Ton
And "Aha!". Well done :-)
I tried the commands and received "permission denied" on the second python instruction "pijuice=..." so I ran python as sudo and there was no error. So I amended the desktop shortcut for the pijuice gui to run as sudo and hey presto! it works and is seeing the battery status changing.
Not sure whether there are implications running as sudo - aside from the obvious security stuff. Presumably there are some file permissions that ought to be sorted out?
Thanks, David

notGMman, 2018-05-04:
Hi Ton
It looks as if it's as simple as adding the user to the "i2c" group. Darn, I shouldn't have fallen for that one. Sorry to have wasted your time. I guess most people are using the "pi" user.
I don't have an icon in the tray - but I have a gap that I can hover over that shows the battery level so I'll work on that!
Regards, David

tvoverbeek, 2018-05-04:
@notGMman You are aware the tray app currently only works when power is connected to the PiJuice USB connector.
The fix is already on github in my PR #94.
In /usr/bin/pijuice_tray.py change `str((b_level/10)*10)` to `str((b_level//10)*10)` on lines 157 and 159.
(source is python3 compatible, single / is floating point division, double // is integer division which is what is needed here)

tvoverbeek, 2018-05-05:
Closing issue

closed 2018-05-05 by tvoverbeek