PiSupply/PiJuice #111: Battery at 0% after about 18+ hours of charging

cmsimike, 2018-05-02:
Hi,

I am following up from: https://twitter.com/PiSupply/status/991685064302977029, additionally ticket #41852 is mine in your system, but I am happy to continue the discussion here.

I got to installing my pijuice today and I don't believe the battery is charging, despite it being enabled in the UI. 0% all day. Any thoughts? I don't want to test it by disconnecting the power, since I don't feel confident from what I see in the ui.

Also tested it out via API call:
[image]

Definitely out of options here. Not sure if the unit is just broken or I'm doing something wrong. As noted in the email I sent, I didn't have install instructions and your learn.* site was down, so I didn't know what to do with the pushrod that came in the bag with the screws, so I didn't use that.

cmsimike, 2018-05-02:
This is the link that I suspect has info about what to do with the pullrod, but it doesn't work:
https://learn.pi-supply.com/make/pijuice-quick-start-guide-faq/

dtwilliamson, 2018-05-03:
Did you make sure to remove the clear plastic tab between the battery terminals and the PiJuice terminals?

cmsimike, 2018-05-04:
@dtwilliamson Thank you for your comment! 

The PiJuice showed up fully assembled, so I assumed it was ready to be put on my Pi and I did.  The short of it is that I didn't remove anything - I just plopped it directly onto my Pi.

Are you suggesting that I can remove the battery?

cmsimike, 2018-05-04:
OMG THAT WAS IT!!! I didn't even notice the plastic strip there
@dtwilliamson Thank you SO MUCH for that. I wouldn't have thought to look.

closed 2018-05-04 by cmsimike