PiSupply/PiJuice #112: apt-get repo address?

winstonford, 2018-05-04:
what is the apt-get repo address for pijuice-base?

tvoverbeek, 2018-05-05:
The pijuice packages are in the raspberry pi specific repo: http://archive.raspberrypi.org/debian/
which should be in /etc/apt/sources.list.d/raspi.list

closed 2018-05-05 by tvoverbeek

winstonford, 2018-05-05:
On my pimusicbox, which is based on debian wheezy, the repo is not found.  On my pis running debian 8 jesse and 9 stretch, apt-get does find and instal pijuice-base.

tvoverbeek, 2018-05-05:
wheezy is too old.
For jessie and stretch do `sudo apt-get update` first before trying to install pijuice-base.

jimklimov, 2022-01-21:
...And bullseye is too new ;(

http://archive.raspberrypi.org/debian/dists/bullseye/main/binary-arm64/Packages lacks pijuice* packages, while http://archive.raspberrypi.org/debian/dists/buster/main/binary-arm64/Packages has them.

(Sorry for the bump, thanks for the URL)

tvoverbeek, 2022-01-22:
If you want to run the packages on 64-bit use multi-arch.
```
sudo dpkg add-architecture armhf
sudo apt-get update
sudo apt-get install libc6:armhf
```
This will install the 32-bit run time libraries, so the 32-bit executables will work on a 64-bit OS.
Note: the older buster 64-bit raspberry pi os already included multi-arch, the bullseye version does not.

karsten-aichholz, 2022-02-03:
> If you want to run the packages on 64-bit use multi-arch.
>
> ```
> sudo dpkg add-architecture armhf
> sudo apt-get update
> sudo apt-get install libc6:armhf
> ```
>
> This will install the 32-bit run time [...]

I believe this should read:
`sudo dpkg --add-architecture armhf`

tvoverbeek, 2022-02-03:
@karsten-aichholz Yes, you are right, my bad.