PiSupply/PiJuice #113: How to turn on (and charge) pijuice so that pi will run when usb power is disconnected?

winstonford, 2018-05-05:
The [manual](https://github.com/PiSupply/PiJuice/blob/master/Documentation/PiJuice%20Guide.pdf) says in step #4 "Switch on the Pi Juice"

How?

Running Raspbian Stretch on pi3.  Hat connected.  Usb power connected to pi3. 

Did:

`apt-get install pijuice-base`

and then:

`service pijuice status`

returns: 
```
● pijuice.service - PiJuice status service
   Loaded: loaded (/lib/systemd/system/pijuice.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2018-05-05 19:03:50 UTC; 10s ago
 Main PID: 705 (pijuice_sys.py)
   CGroup: /docker/dd0db41950ef8d2807fd8145f72ca6b05837eafccec2cee5a7442c1051bf23cc/system.slice/pijuice.service
           └─705 /usr/bin/python /usr/bin/pijuice_sys.py

```

On pijuice hat led is blinking, alternating green and blue. Red led will periodically come on for a few seconds and go off.  On pi, led is solid red.

Usb power is plugged into pi. Is this correct? 
When I pull the plug to disconnect usb power, all leds go dark, pi loses power.  

High build quality on the hardware.  Nice work!  Looking forward to getting the pi on the juice.

shawaj, 2018-05-07:
If you have a red light on the Pi then it's getting power fine. Have you checked the OS?

winstonford, 2018-05-07:
Yes, power to the pi is fine.  As you can see above , the pi is online and I can ssh in and run the commands to install and run your app.

The os is rasbian 9.

My original questions:

1. The manual says in step #4 "Switch on the Pi Juice" How?
2. Usb power is plugged into pi. Is this correct?

The Problem: When I pull the plug to disconnect usb power to the pi, the pi loses power; it should remain on, powered by pijuice.

tvoverbeek, 2018-05-07:
Your PiJuice is already on. The alternating green/blue LED means the PiJuice battery is charging (blue) and has a charge of > 50% (green).
USB connected to RPi is fine. However in pijuice_cli check in General Settings that 'GPIO input enable' and 'Charging Enabled' are selected. In your case they should be since you have a green/blue blinking LED.
Normally the Pi should keep power when you pull the power.
First time I tried, the red Pi LED became weaker and the Pi stopped.
Could not repeat it later.
Suggest to let your battery charge a while and try again.
Did you update the firmware to 1.1?

winstonford, 2018-05-07:
1. Thank you for the info on led color and blink patterns.

2. pijuice_cli.py shows:
[X] GPIO input enable 
[X] Charging enabled

3. Yes I have waited hours and tried with same result, pi loses power.
4. I did not update firmware, but cli shows it is current version is 1.1 and status is up to date, although I see here that latest version is 1.2: https://github.com/PiSupply/PiJuice/tree/master/Firmware

5. in pijuice_cli.py under the first option: status, the battery cycles between 0% and 95% and battery voltage fluctuates, where as input voltage remains constant.  please see attachments.  

6. Is this normal for battery to fluctuate from 0-95%?

[image]
[image]
[image]
[image]
[image]

tvoverbeek, 2018-05-07:
The switching between 0% and 95% is not normal.
Maybe cleaning the battery contacts on both the PiJuice and the BP7X battery might help.

tvoverbeek, 2018-05-07:
Could you also try with the power connected to the PiJuice USB?

winstonford, 2018-05-07:
Initially, I did plug usb power directly into the pi juice, when I saw no response, I tried plugging into the pi, which powered up both (connected) units.

Just now I unplugged power to the pi and plugged it into the pi juice.  After a moment, a red led came on briefly, went off, and leds stayed dark.  Pi did not power on.

I unplugged power, removed the battery. The terminals on battery and juice are clean and clearly new, but I cleaned them with alcohol, and replaced battery.  The red led on the juice came on momentarily and then stayed off.

Now when I plug directly into the pi juice, the leds come on and alternate between green and blue, indicating charging and charge state as you mentioned above.  The pi does not power on.  I will leave the unit as it for a few hours to see if it may charge fully.

1. What led color pattern indicates battery full?
2. When the pi juice is directly plugged into usb power, should it automatically power on the pi?

winstonford, 2018-05-07:
Update:

1. **The pi stays on when power is disconnected, so my initial problem in solved.** 
- The juice has a solid green led, and the pi is mostly dark.  Periodically, the pi blinks a green led 3 times.
- I can plug into either pi or juice for power, and on refresh the status screen will show the change of charging source is correctly detected.

2.  **I believe I may have a clue as to why the unit was not functioning properly.**  While swapping out power sources and refreshing the pijuice_cli.py status screen, I noticed that when unplugged, the battery charge state is 87%, yet when plugged in, it jumps to 100%.   The battery voltage also jumps.  This false reading occurs when you plug in power to either the pi or the juice.

Typically, when you read a battery voltage, you want it disconnected from power.  Leave it disconnected for a period, and then measure.  It appears the unit is measuring battery voltage *while* it is connected to power.

So if juice battery power is low upon arrival to the customer, he attaches the unit to the pi, powers up the rig, and the voltage jumps.  So the unit reads that the battery is full, and does not charge.

I just opened my second juice and before connecting I am charging.  I will see if this may produce different results.  I will clean the contacts with alcohol prior inserting the battery and report back.

So to fix the problem, the firmware might be updated to disconnect power from charging source, wait x seconds, then measure battery voltage.  And the short term work around may be to charge the juice battery fully prior to connecting juice to pi.

KeithSloan, 2018-05-08:
But would the miss reading of the state of the battery explain the pi losing power? I would have thought the pi-juice hardware would just supply what battery power there was without bothering to check the battery level. Checking the state of the battery takes time and although it is only a short time I would have thought it was long enough that the pi would experience a power failure.

winstonford, 2018-05-08:
Yes, the pi juice would supply what power there is, but if the charge state is low, the voltage is low, and if it's low enough, the pi won't have sufficient power to stay on or boot when the input usb power is removed.

Checking the charge state of the battery does take a short time, and it's true that it would be most accurate to disconnect everything, but I would only disconnect the battery from the usb charging input, and keep the pi drawing from the battery.  The pi draw is constant enough that you could compensate when calculating charge state with it remaining connected.  As for peripherals connected to the pi or juice and drawing power, that might be less constant and more challenging to compensate for in calculating charge state.

KeithSloan, 2018-05-08:
Well, I fail to see how what you changed fixed the problem of the pi losing power.

winstonford, 2018-05-08:
Quick update: 

The seconds pijuice is up and running smoothly after pre-charging prior and cleaning terminals prior to connecting to the pi.

The first pijuice, however is throwing an error.  I issued a shutdown now cmd but it would not power down.  So I removed the battery and replaced it to power cycle.  After reboot, I issue it returns:

       root@afeb57b:~# pijuice_cli.py

It returns:

Traceback (most recent call last):
  File "/usr/bin/pijuice_cli.py", line 1229, in <module>
    urwid.MainLoop(top, palette=[('reversed', 'standout', '')]).run()
  File "/usr/lib/python2.7/dist-packages/urwid/main_loop.py", line 278, in run
    self._run()
  File "/usr/lib/python2.7/dist-packages/urwid/main_loop.py", line 376, in _run
    self.event_loop.run()
  File "/usr/lib/python2.7/dist-packages/urwid/main_loop.py", line 682, in run
    self._loop()
  File "/usr/lib/python2.7/dist-packages/urwid/main_loop.py", line 719, in _loop
    self._watch_files[fd]()
  File "/usr/lib/python2.7/dist-packages/urwid/raw_display.py", line 393, in <lambda>
    event_loop, callback, self.get_available_raw_input())
  File "/usr/lib/python2.7/dist-packages/urwid/raw_display.py", line 493, in parse_input
    callback(processed, processed_codes)
  File "/usr/lib/python2.7/dist-packages/urwid/main_loop.py", line 403, in _update
    self.process_input(keys)
  File "/usr/lib/python2.7/dist-packages/urwid/main_loop.py", line 503, in process_input
    k = self._topmost_widget.keypress(self.screen_size, k)
  File "/usr/lib/python2.7/dist-packages/urwid/container.py", line 592, in keypress
    *self.calculate_padding_filler(size, True)), key)
  File "/usr/lib/python2.7/dist-packages/urwid/decoration.py", line 621, in keypress
    return self._original_widget.keypress(maxvals, key)
  File "/usr/lib/python2.7/dist-packages/urwid/listbox.py", line 987, in keypress
    key = focus_widget.keypress((maxcol,),key)
  File "/usr/lib/python2.7/dist-packages/urwid/wimp.py", line 535, in keypress
    self._emit('click')
  File "/usr/lib/python2.7/dist-packages/urwid/widget.py", line 464, in _emit
    signals.emit_signal(self, name, self, *args)
  File "/usr/lib/python2.7/dist-packages/urwid/signals.py", line 264, in emit
    result |= self._call_callback(callback, user_arg, user_args, args)
  File "/usr/lib/python2.7/dist-packages/urwid/signals.py", line 294, in _call_callback
    return bool(callback(*args_to_pass))
  File "/usr/bin/pijuice_cli.py", line 1191, in item_chosen
    callback()
  File "/usr/bin/pijuice_cli.py", line 62, in __init__
    self.main()
  File "/usr/bin/pijuice_cli.py", line 109, in main
    status_args = self.get_status()
  File "/usr/bin/pijuice_cli.py", line 80, in get_status
    gpio_info = "%.3fV, %.3fA, %s" % (io_voltage / 1000, io_current / 1000, status['powerInput5vIo'])
KeyError: 'powerInput5vIo
root@afeb57b:~# 

Which brings  me to my next question which may well warrant posting a new issue:

What is best practice for powering down the pijuice?

tvoverbeek, 2018-05-08:
You get this error when the PiJuice is in a funny state or when the firmware is damaged.
See the discussion in Issue #105.
Try the recovery procedure outlined in my comment from yesterday: https://github.com/PiSupply/PiJuice/issues/105#issuecomment-387147908

ChristopherRush, 2019-09-15:
Solution here - https://github.com/PiSupply/PiJuice/issues/113#issuecomment-387380270

closed 2019-09-15 by ChristopherRush