PiSupply/PiJuice #114: Documentation is hopelessly inadequate.

KeithSloan, 2018-05-06:
I feel I am missing a source or sources of information. The user guide does not tell me what I need to know. Like what is the pogo pin for? Where does it go?

Need to fully understand the implications of plugging supply into pi versus into pijuice.

Explanation of output from GUI and what some of the terms mean.

KeithSloan, 2018-05-06:
PIjuice GUI settings

What does system switch off | 500ma | 2100ma mean?
My GPIO input toggles between PRESENT, WEAK and BAD what is that all about?
What does -ve current on GPIO mean?

shawaj, 2018-05-06:
Have you read the hardware readme on GitHub?

https://github.com/PiSupply/PiJuice/blob/master/Hardware/README.md

KeithSloan, 2018-05-06:
Have scan read and rescanned just now. But do not see where it answers any of the questions

shawaj, 2018-05-06:
It answers most if not all of your questions on the various GitHub readmes

Madko, 2018-05-06:
And what does "System power switch" mean ? (off|500mA|2100mA)

ChristopherRush, 2018-05-10:
This is with regards to the VSYS output on header P3. It is documented under https://github.com/PiSupply/PiJuice/blob/master/Hardware/README.md

mwrich4, 2018-05-17:
I keep allowing the arrogance of this group to amaze me.  

Instead of posting an answer for others to find, we get the SAME "I'm always right" replies that have plagued the communications of this project.

shawaj, 2018-05-17:
@mwrich4 - is there something you need help with at all?

mwrich4, 2018-05-18:
I mentioned the inadequate documentation before you started shipping and before you tore down the website.  I think the current state of things illustrates the issues.  (off topic:  Pie-Hole ?!   REALLY ??)

shawaj, 2018-05-18:
@mwrich4 - I asked before...is there something you need help with at all? Because you are saying the documentation is inadequate but you aren't actually saying what you need help with at all?

And for the record, the vast majority of the 2400+ people who 3500+ units that we've shipped to have absolutely no issue whatsoever with either the documentation or the product itself. I think our customer service and our assistance here speaks for itself so I won't say anything more than that.

shawaj, 2018-05-18:
@tvoverbeek @KeithSloan agreed, someone else mentioned on another thread having tool tips next to each item with a bit of explanation of what it is/does. Think this is a really good idea and we are progressing it but it'll take some time. Thanks for the suggestion though

Madko, 2018-05-18:
How do we set event triggers using the CLI ? I can find tips in the documentation but only with the GUI.

shawaj, 2018-05-18:
@Madko please start a new issue rather than hijacking this one. It'll be easier for everyone now and in the future to have the answer in a dedicated issue

Madko, 2018-05-18:
Ok I will start a new issue but I was not hijacking this one (I almost started it), it's still about inadequate documentation. And since you were asking if someone needed help. But Ok no big deal :)

shawaj, 2018-05-18:
@Madko didn't mean hijacking in a bad way it's just easier to have everyone's individual issues separately so we can keep track of things better. Thanks very much! 😊

klint-k, 2018-05-21:
Pogo??? #10

shawaj, 2018-07-19:
FYI the maker zone links should be back up now

ChristopherRush, 2019-09-15:
I believe everything is now adequately documented, if you do not agree then reopen.

closed 2019-09-15 by ChristopherRush