PiSupply/PiJuice #115: AttributeError: 'WakeupAlarmTab' object has no attribute 'status_text'

Madko, 2018-05-06:
Hi,

When I run pijuice_cli.py and try to change the wake up alarm (because the one I previously set doesn't work), I've got this crash:
``Traceback (most recent call last):
  File "/usr/bin/pijuice_cli.py", line 1229, in <module>
    urwid.MainLoop(top, palette=[('reversed', 'standout', '')]).run()
  File "/usr/lib/python2.7/dist-packages/urwid/main_loop.py", line 278, in run
    self._run()
  File "/usr/lib/python2.7/dist-packages/urwid/main_loop.py", line 376, in _run
    self.event_loop.run()
  File "/usr/lib/python2.7/dist-packages/urwid/main_loop.py", line 682, in run
    self._loop()
  File "/usr/lib/python2.7/dist-packages/urwid/main_loop.py", line 719, in _loop
    self._watch_files[fd]()
  File "/usr/lib/python2.7/dist-packages/urwid/raw_display.py", line 393, in <lambda>
    event_loop, callback, self.get_available_raw_input())
  File "/usr/lib/python2.7/dist-packages/urwid/raw_display.py", line 493, in parse_input
    callback(processed, processed_codes)
  File "/usr/lib/python2.7/dist-packages/urwid/main_loop.py", line 403, in _update
    self.process_input(keys)
  File "/usr/lib/python2.7/dist-packages/urwid/main_loop.py", line 503, in process_input
    k = self._topmost_widget.keypress(self.screen_size, k)
  File "/usr/lib/python2.7/dist-packages/urwid/container.py", line 592, in keypress
    *self.calculate_padding_filler(size, True)), key)
  File "/usr/lib/python2.7/dist-packages/urwid/decoration.py", line 621, in keypress
    return self._original_widget.keypress(maxvals, key)
  File "/usr/lib/python2.7/dist-packages/urwid/listbox.py", line 987, in keypress
    key = focus_widget.keypress((maxcol,),key)
  File "/usr/lib/python2.7/dist-packages/urwid/wimp.py", line 535, in keypress
    self._emit('click')
  File "/usr/lib/python2.7/dist-packages/urwid/widget.py", line 464, in _emit
    signals.emit_signal(self, name, self, *args)
  File "/usr/lib/python2.7/dist-packages/urwid/signals.py", line 264, in emit
    result |= self._call_callback(callback, user_arg, user_args, args)
  File "/usr/lib/python2.7/dist-packages/urwid/signals.py", line 294, in _call_callback
    return bool(callback(*args_to_pass))
  File "/usr/bin/pijuice_cli.py", line 1191, in item_chosen
    callback()
  File "/usr/bin/pijuice_cli.py", line 983, in __init__
    self.device_time = self._get_device_time()
  File "/usr/bin/pijuice_cli.py", line 1052, in _get_device_time
    str(t['second']).rjust(2, '0')))
  File "/usr/bin/pijuice_cli.py", line 1058, in _update_status
    self.status_text.set_text("Status: " + self.status)
AttributeError: 'WakeupAlarmTab' object has no attribute 'status_text'
``
Any idea why ?

I'm using pijuice-base 1.3.2-1 on raspbian up-to-date.

Best regards,
Edouard

tvoverbeek, 2018-05-06:
Yes. Already fixed on github. I just merged the fix (and lots of others) with the master branch.
Replace /usr/bin/pijuice_cli/py with the one from github (https://github.com/PiSupply/PiJuice/blob/master/Software/Source/src/pijuice_cli.py).
While you are at it, you might as well replace /usr/lib/python2.7/dist-packages/pijuice.py, /usr/bin/pijuice_tray.py, /usr/bin/pijuice_gui.py with the github versions to get all the fixes.

Madko, 2018-05-06:
any idea when the deb package will include those fixes ?

tvoverbeek, 2018-05-06:
No. I am in contact with Pi Supply and try to push them to update the deb packages asap.

ilyastrodubtsev, 2018-05-21:
In the version that is on the githab, this bug is fixed.

closed 2018-05-21 by ilyastrodubtsev

Madko, 2018-05-22:
Is that fixed in the .deb package from raspbian repo ?

shawaj, 2018-05-22:
Yes v1.3.3

Madko, 2018-05-22:
Great ! Thanks to you all for this fix :)