PiSupply/PiJuice #116: Links are down for maintenance for days (even weeks)

Madko, 2018-05-06:
Hi,

All the links in the doc and from pi supply point out on a dead end or back to pi supply shop.

https://learn.pi-supply.com/make/pijuice-quick-start-guide-faq/ => down for maintenance
same for:
http://pisupp.ly/make
http://pisupp.ly/remote-camera => shop pisupply
http://pisupp.ly/weather-station => shop pisupply
http://pisupp.ly/robot => shop pisupply

All those pisupp.ly redirect to learn.pi-supply.com

Since this is the case for weeks, and that I don't find any report here about that problem, I'm opening this issue. I hope this is helpful.

Best regards,
Edouard

rmiles7721, 2018-05-07:
I've been trying to reach pisupply for several weeks as well with the same problems. Finally found directions on the web to put together the brightpi I purchased. Same thing with my Pijuice.

ChristopherRush, 2018-05-08:
We are aware of the issue regarding the learn.pi-supply.com URLs and we are working hard to gets these up and running ASAP.

Madko, 2018-05-21:
This is still not working. Do you have alternative links ? I'm specially interested in the remote camera and the weather station.

Bigtweety, 2018-06-28:
Hi, indeed all pages on this site are still down.
1. Why would the site owner use such a misleading message when (after almost 2 months) we can clearly see the issue seems much more serious than maintenance. To me, it looks very much like their website manager has left with possibly all its contents. 
2. With hardware still getting sold and pointing to many pages on this website, the least that could be done would be to offer an alternate way to get the info. 
3. Does anyone have links to other pages with the same/similar info? I'm looking specifically for the pijuice related stuff.
4. If this is problematic for you, do as I did, contact them on sales@pi-supply.com to ask for the info and an expect re-opening of the website. 

Cheers
BT

ChristopherRush, 2018-06-28:
@Bigtweety What in particular are you looking for with regards to the PiJuice? Almost everything is documented on our GitHub PiJucie pages.

Bigtweety, 2018-06-28:
Here is what I received after sending that email to them. [
Getting Started with PiJuice-2.pdf](https://github.com/PiSupply/PiJuice/files/2145393/Getting.Started.with.PiJuice-2.pdf)

They ignored my question about the web-site, only to say that https://github.com/PiSupply/PiJuice is to be used to look for pijuice related stuff for now.... 

Hoping that helps others
BT

ChristopherRush, 2018-06-28:
@Bigtweety Unfortunately that is correct. We simply do not have a date as to when the Maker zone will be back up and running, there are a number of unknown issues we are trying to resolve. As everything is currently documented on GitHub and with other projects going on, this is simply not at the top of our list. However, please feel free to fire away any questions or queries you have relating to the PiJuice via email or raise an issue on GitHub.

shawaj, 2018-07-19:
Maker zone is back up now. Will close this issue but let us know if you find any dead links

Madko, 2018-07-20:
Almost fixed, everything works except :

http://pisupp.ly/remote-camera => shop pisupply
http://pisupp.ly/weather-station => shop pisupply
http://pisupp.ly/robot => shop pisupply

shawaj, 2018-07-20:
@Madko those guides do not exist yet.

@ChristopherRush is working on them. I will re-open this and then we can close once those are available

shawaj, 2020-02-22:
Closing as these are all now available

closed 2020-02-22 by shawaj