PiSupply/PiJuice #117: Pi Juice dont charge. Batocera

toralora, 2018-05-07:
Hi 
Ive got a problem  i wanna make a portable Console with Batocera and bought Pijuice.  but it doesnt charge and ive read that  Batocera/Recalbox are not able to  add other Programs. Is there a option to use both?

tvoverbeek, 2018-05-07:
Suggest you first install Raspbian (maybe on a separate SD card) and install the pijuice packages.
Check the configuration and make sure the 'Charging Enabled' check box is selected on the General tab in the PiJuice HAT Configuration. Configure other parameters of the PiJuice as you need it.
When satisfied install Batocera (or switch SD cards).

shawaj, 2018-05-14:
Also - have you made sure to remove the plastic battery tab between the battery and battery contacts?

toralora, 2018-05-14:
the plastic tab ^^  that was never the Problem.  ive Tried  with Raspbian  and with  onetime Remove the Battery completly it worked. thx

closed 2018-05-14 by toralora