PiSupply/PiJuice #118: Please emphasize need to remove clear plastic tab before use

claremacrae, 2018-05-13:
As per #111 it's easy to miss the need to remove the plastic tab on first use.

I had a go at creating a pull-request to improve the docs, but somehow it ended up on my clone instead of here.

Can anyone more familiar with github merge https://github.com/claremacrae/PiJuice/pull/1 into this repo please? Or write some similar wording?

Thanks.

grappler, 2018-05-14:
@claremacrae I hope this documentation helps. https://help.github.com/articles/creating-a-pull-request-from-a-fork/

Instead of making a pull request to your own repo/fork you need to make the pull request to this repo.

shawaj, 2018-05-14:
Thanks @claremacrae will add your changes later on today

shawaj, 2018-05-14:
Done - https://github.com/PiSupply/PiJuice/commit/697c55813806dda66589a1525dd834938d9c590f

closed 2018-05-14 by shawaj