PiSupply/PiJuice #119: Translucent battery mounting clip on PiJuice blocks HATs

claremacrae, 2018-05-15:
This picture shows how the plastic clip that holds the supplied BP7X battery in place on my PiJuice is preventing HATs from being used.

[image]

I've tested this with the following, and they all have the same problem:

* Displayotron
* Explorer HAT Pro
* Sense HAT gets a bit closer, but not enough - gets power, to show rainbow pattern, but the pattern does not disappear - the behaviour is as described in https://github.com/RPi-Distro/python-sense-hat/issues/70

An Explorer PHAT does work, though.

I presume I'm going to have to trim back the plastic clip, but I wanted first to know how others had solved this problem, or if there us something unusual about the plastic clip on my PiJuice?

ChristopherRush, 2018-05-15:
>    described in RPi-Distro/python-sense-hat#70

I've had no issues with any of the HATs I've tried. Have you tried to
loosen the screw a bit? This might give the plastic a bit of leeway.

ChristopherRush, 2018-05-16:
Here is how one of my HATs fit

[image]

It doesn't sit all the way down because of the solder blobs, but it does make good contact. I would also say that your header looks much shorter, therefore will not sit flush with the PiJuice board. I'm assuming all these boards mentioned above still work?

shawaj, 2018-05-16:
We have spares of the battery surrounds if your one can't be moved into position @claremacrae but as Chris has said you should be able to move it

claremacrae, 2018-05-16:
So I think part of the issue may be that some HATs have much greater depth of plastic than others around the GPIO pins. Where a HAT has very shallow black plastic around the GPIO, it’s possible that circuitry on the underside of the HAT can clash against the bits and pieces on the top of the PiJuice.

@ChristopherRush what HAT is in your photo?

Ah - I see I have just stated exactly what you said, @ChristopherRush - thanks.

So it may be that I need to get some kind of male-to-female HAT extended, to allow the HAT to sit something like 1cm higher away from the PiJuice and then it will all work...

claremacrae, 2018-05-16:
I’ll do some more experiments tomorrow....

shawaj, 2018-05-16:
Yes we are actually selling some headers like that on our shop for that very reason.

Guess we need to make a note about that on here somewhere

Any testing you can do for clashes would be fantastic. We should maybe start a list here

claremacrae, 2018-05-16:
Thanks @shawaj - is this the product you mentioned?

https://uk.pi-supply.com/products/40-pin-female-stacking-header

shawaj, 2018-05-16:
You beat me to linking you. Yes that's it. I think we have some different height ones too but those work great

shawaj, 2018-05-16:
@claremacrae do you mind if I use your picture above to explain this on the readme?

claremacrae, 2018-05-16:
@shawaj Yes, no problem using the pic - thanks for asking.

claremacrae, 2018-05-16:
I guess the thing is that once someone uses one of those stacking headers, they will then have a different problem that any stand-offs that they had previously used with the HAT will now not be high enough?

So then they’d either need to be really careful to not bend/break the PHAT or pins on the Pi, or source extra stand-offs.

 So I think that using PiJuice with (some) HATs is looking a bit challenging right now?

claremacrae, 2018-05-16:
I ordered and just received “Adafruit Black Nylon Screw and Stand-off Set – M2.5 Thread” - the cost is all adding up though, and you have to know these things exist in the first place....

claremacrae, 2018-05-17:
OK, I think I may be wrong on this one.

The script I was using to detect if the HAT was connecting no longer works - so it's possible - or probable - that some of these HATs were connected after all...

I've logged the problem with my script here: https://github.com/claremacrae/raspi_hat_data/issues/1

shawaj, 2018-05-20:
Thanks. We will get a note about this added to the documentation

shawaj, 2018-07-25:
@ChristopherRush can you document anything here that isn't already and then close?

ChristopherRush, 2019-09-15:
Potential solution here - https://github.com/PiSupply/PiJuice/issues/119#issuecomment-389632660

closed 2019-09-15 by ChristopherRush