PiSupply/PiJuice #120: Battery only last around ½ hour.

joda58, 2018-05-17:
I do not know if I installed it the correct way (documentation could be better), 
but I **ONLY get around ½ hours work** after 24 hours of charging.
I am running on a Pi  3B+.

When I look at the GUI the percentage % (sometimes) jumps up and down. 

Did I get a bad battery or did I configure something wrong?
How much battery time do you get?
How much should I get?

This is my longest kickstarter,
/JD

shawaj, 2018-05-17:
What's running on the pi when you get that? It should be about 3 to 4 hours on a fully charged pi 3b+ but it depends what is running

joda58, 2018-05-17:
Sorry wrong button :-)

ChristopherRush, 2018-05-17:
What you should really be looking at here is the overall current consumption.

jdotbdot, 2018-05-20:
[image]

This is a Pi 3B+ at 100% CPU utilisation, no HDMI or keyboard/mouse.
1hr 13 minutes from fully charged to shutdown

joda58, 2018-05-27:
I ran my Pi3B+ without HDMI, network and keyboard/mouse and I got 1½ hours with "no" CPU load.
Have I got a faulty PiJuice battery? (Firmware 1.2)

/JD

ChristopherRush, 2018-05-27:
What's the load like?

ilyastrodubtsev, 2018-06-01:
It seems that you have problems with the battery if the processor was not loaded 100%

joda58, 2018-06-03:
Ok, If I have a problem with the brand new battery ... how do I get it changed!!!!!

/Jonas D.

tvoverbeek, 2018-06-03:
1.5 hours seems reasonable for a Pi3B+

What Raspbian version are you running?
When full, the desktop is running even when no HDMI connected, WiFi is on and probably Bluetooth too. The new ethernet chip on the 3B+ uses much more power than the previous 3B.
All in all the 3B+ uses about 170mA more than the 3B.
My 3B+ when idling on the desktop, WiFi on uses between 700 and 800 mA.
With the 1800mAh BP7X this gives 1.5 hours for 750mA

So measure your actual current use when idling as @ChristopherRush suggested above.
If you want a longer duration use a less power hungry RPi or get a larger battery.
See e.g. http://raspi.tv/2018/how-much-power-does-raspberry-pi-3b-use-power-measurements
and the calculator https://github.com/PiSupply/PiJuice/blob/master/Hardware/Batteries/PiJuice%20Battery%20Discharge%20Levels.xlsx

closed 2019-09-15 by ChristopherRush