PiSupply/PiJuice #121: Some observations and comments

github-0, 2018-05-17:
Hi, got my Pijuice couple days ago (thanks!) and after playing with it for a while here are some observations I made:

1. If the battery level is already below min_charge threshold and system is booted, it will be shut down forcefully (power is cut off). Some while after starting the system I can see LED2 flash like it does when shutdown is initiated but then after a moment power is cut off while the Pi is still working either with the bootup or with the shutdown sequence (Pi's activity led has been blinking since powering it on so can't really tell). Dirty shutdown confirmed through syslog ("Dirty bit is set. Fs was not properly unmounted and some data may be corrupt"). I'm using SYS_FUNC_HALT_POW_OFF as the low charge function. Seems reproducible.

2. Is the blinking blue led indicator (charging) supposed to never stop blinking? Hardware documentation states "For full buttery state blue component is steady on". This never happens in my case. When the battery is full (was connected to power for 12+ hours) the battery level never reaches 100 but fluctuates between 97-98.

3. Related to that, are the default LED1 battery level color indicators user configurable? Would solve my problem by showing steady blue (or shut off) when charge level is over 95 for example. 

4. I see that the default is to not initiate a safe shutdown when battery is empty. Wouldn't it be better to have "safe" defaults? At least warn about this in the documentation please, I had to learn the hard way :)

5. Did a resilience test to see how long the battery (BP7X) lasts with my Pi (wifi and Z-Wave radios connected):

down to 50% in 1 hour
down to 25% in 17 min
down to 0% in 2h

comparing that to the test posted here ([https://github.com/PiSupply/PiJuice/blob/master/Hardware/Batteries/pijuice_charge.log](https://github.com/PiSupply/PiJuice/blob/master/Hardware/Batteries/pijuice_charge.log)):

down to 50% in 1h 25min
down to 25% in 30 min
down to 0% in 2,5h

Seems like the last 25% lasts longer than the first 75%. Documentation says discharge is "pretty linear" but I wouldn't be so sure :). Maybe there'll be some tweaks or a minute estimation added to the software in the long run but in the mean time maybe this could be added to the documentation as a general idea of what to expect?

ChristopherRush, 2018-05-17:
1. I would imagine your min_charge threshold is set too low. The Pi triggers the shutdown procedure but then your current is high and the Pi looses power and cuts off.

2. If your Raspberry Pi is on and drawing current then it is highly unlikely that the battery will reach 100% hence the flashing blue LED.

3. Nope, not needed as it does the job by displaying the battery level and charging status.

4. It does what every other battery/UPS would do when it reaches zero. If you wish to configure it, then use the software provided.

5. We are aware of how the battery reports its percentage levels and due to the characteristic of the battery it is not possible to change this at the moment. Battery percentage drops as current rises and vice versa, this only effects the discharge levels not charge.

github-0, 2018-05-17:
1. min_charge was set to 90 during testing.

2. Seems logical.

3. Yes but only that the battery level is between 50 and 100 percent. If I could configure a threshold to shut off the blinking when charge level is over 95 I could get even more information out of the leds (although considering nr 5, maybe this kind of precision is irrelevant).

4. Fair enough. Still, I fail to see the benefit of abruptly losing power vs. a safe shutdown.

5. I could be misinterpreting this but isn't it the other way around? From the docs "there are measuring errors especially while charging because there are big currents over 1 Amp". This is also what I see when plugging in a power source -> charge level makes a big jump up. Anyway, not a big issue. I just found it interesting to know that while the charge level shows 25% I still have over 75% of total capacity left.

github-0, 2018-05-18:
Sorry, I should have clarified that this is after the software is installed, ie. I have to enable the safe shutdown myself, it is not on by default. This is at least the case with the light version with no GUI.

ChristopherRush, 2018-05-18:
The default software settings are almost like a blank canvas, every user will have their own agenda to configure the PiJuice software for there projects. Once we start changing the default settings, it will not always be suited for every user. Don't forget there are a number of different type of shutdowns the Raspberry Pi can do. I fully understand where you are coming from but for us it makes sense to let the users configure their board the way they want. Maybe we can update the documentation to reflect this point further after installing the software.

closed 2019-09-15 by ChristopherRush