PiSupply/PiJuice #123: TUI is missing events trigger configuration

Madko, 2018-05-18:
As @shawaj suggested I'm opening this new issue:

I can't find a way to configure events trigger using the TUI. Can't find any pointer in the documentation neither. I'd like to shutdown properly the OS and the PI when battery level is less than 15%. And power it back when battery level is above 50%.
The only thing I can find in the doc is that "using the light version of PiJuice" we can directly edit a json file. What does light version of Pijuice mean ? Is that the TUI ?
Due to lack of documentation on this matter, and the fact that for preservative power project you can't use the GUI/X11 etc, I tried to modify this json file, but now my pi just poweroff just after booting... which means that any help would be greatly appreciated. Thanks

Best regards,
Edouard

tvoverbeek, 2018-05-18:
The 'light' version is what you get with the pijuice-base package: The pijuice module, the pijuice service (pijuice_sys.py) and the CLI (or TUI) pijuice_cli.py.
Yes, we know pijuice_cli does not have all the functions from the gui (such as the event trigger and the user functions to hook up to certain events).
I am sure these will be added for the next version, but for now you will have to manually edit the json file (and yes I know, the format is not documented)

tvoverbeek, 2018-05-18:
The pijuice-base package installation creates the (alsmost empty) file.
You will find it in /var/lib/pijuice/pijuice_config.JSON.
When you change a setting (in the gui) the file gets updated.
The current pijuice_cli does not change any settings in the JSON file.

Madko, 2018-05-18:
Thanks @tvoverbeek this will be very helpful
But I'm curious then, where are stored the changes made by pijuice_cli ?

tvoverbeek, 2018-05-18:
The changes made by pijuice_cli (and the equivalent ones in pijuice_gui) are all stored in the microprocessor memory on the pijuice board. E.g. led settings, wakeup alarm, button settings

Madko, 2018-05-18:
Oki thank you for all this clarification.

shawaj, 2018-05-18:
@wikipedia555 can we add this to the pijuice_cli the ability to manipulate the JSON?

and @ChristopherRush do you think we can work to document the format of the JSON?

shawaj, 2018-05-18:
@tvoverbeek was there more info you thought would be useful than what is documented her - https://github.com/PiSupply/PiJuice/tree/master/Software#json-configuration-file ?

ChristopherRush, 2018-05-18:
@shawaj I think we can document the JSON file better, I will take a look at this.

shawaj, 2018-05-18:
@tvoverbeek I have asked @wikipedia555 to add this to the cli and then we can document it all better

jdotbdot, 2018-05-22:
On a Pi with just pijuice_base, are configuration changes made via pijuice_cli.py preserved:
If the Pi is rebooted?
If the Pi is shutdown because the battery level reaches the minimum value?
If the battery is removed while the Pi is powered up?
If the battery is removed when there is no USB power?
If the firmware is updated via pijuice_cli.py?

And on a Pi with just pijuice_base, can I configure it by editing pijuice_config.JSON?
Does the Pijuice have to be rebooted to re-read this file? How?