PiSupply/PiJuice #126: Sense Hat stacked onto PiJuice & I2C assignment

sensemonster, 2018-05-21:
My first out of the box attempt at stacking a Sense Hat (https://www.raspberrypi.org/products/sense-hat)  not going well as it appears PiJuice is already using the deafult port.  
Tests show the Sense Hat as "not found".

`i2cdetect -y 1`  shows:
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:              -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- 14 -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --  

Is it possible to change PiJuice's I2C assignment please?

sensemonster, 2018-05-21:
BTW I tried https://github.com/PiSupply/PiJuice/issues/97#issuecomment-383384562
and now see the following from i2cdetect but still unable to "find" the Sense Hat.  I will try direct connection of the Sense Hat and see.

     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- 14 -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --

ChristopherRush, 2018-05-21:
@sensemonster You should be able to change the I2C address for the PiJuice under the GUI menu in PiJuice HAT Configuration > General > I2C Address1

sensemonster, 2018-05-21:
I changed the ID EEPROM address to 52 but that didn't do the trick :-) I will have a go with the i2C Address value in General.

sensemonster, 2018-05-21:
Hmm it appears the I2C address is changed back to 68 after reboot.  The EEPROM value 50 and 52 tried and makes no difference.

sensemonster, 2018-05-21:
i'm trying to set up an off-grid sensor so I am keen to use the Sense Hat with PiJuice :-)

ChristopherRush, 2018-05-21:
68 address is for the RTC 14 is for the PiJuice HAT. I don't have a sense
HAT so I can't really check at the moment.

sensemonster, 2018-05-21:
Ok Christopher thanks for your time anyway - there is no urgency - just irritation :-)

Anyway, I set the EEPROP address to 52 and rebooted.  "No Go" :-)  The strategy was as both PiJuice and the Sense Hat share 50 for the EEPROM then moving PiJuice to 52 would stop the conflict.

`dmesg | grep sense` doesn't reveal anything so I will see if there is anything that can be moved on the Sense Hat addressing.     Puzzling nevertheless

shawaj, 2018-05-21:
@ChristopherRush email David and he'll send you a Sense hat

tvoverbeek, 2018-05-21:
Just tested it with a Sense Hat on top of my PiJuice. Works, but there are a few things to remember:
- There can only be 1 HAT responding on address 0x50 on i2c bus 0.
- Therefore we must change the ID eeprom address from 0x50 to 0x52.
- To change the ID eeprom address the 'ID EEPROM Write unprotect' item has to be selected (!!!)

With only the PiJuice HAT on the default ID EEPROM address 0x50 I get the following
```
pi@rpi-stretch-full:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- 14 -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
pi@rpi-stretch-full:~ $ cat /proc/device-tree/hat/product
PiJuice HATpi@rpi-stretch-full:~ $ 
```
Note the 'UU' for the PiJuice HAT RTC (kernel module for the RTC loaded) and /proc/device-tree/hat/product exists.

Now with PiJuice (still on 0x50) and the Sense Hat:
After boot the Sense Hat LEDs shows a rainbow display.
```
pi@rpi-stretch-full:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- 14 -- -- -- -- -- -- -- 1c -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- 46 -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- 5c -- -- 5f 
60: -- -- -- -- -- -- -- -- 68 -- 6a -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
pi@rpi-stretch-full:~ $ cat /proc/device-tree/hat/product
cat: /proc/device-tree/hat/product: No such file or directory
pi@rpi-stretch-full:~ $ 
```
Note: The PiJuice RTC now shows '68' instead of 'UU' meaning no RTC kernel module loaded.
The I2C addresses for the Sense HAT (0x1c, 0x46, 0x5c, 0x5f, 0x6a) show up.
See also https://pinout.xyz/pinout/sense_hat

After changing the PiJuice HAT to 0x52  (with 'ID EEPROM Write unprotect' selected) and without the Sense HAT:
```
pi@rpi-stretch-full:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- 14 -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
pi@rpi-stretch-full:~ $ cat /proc/device-tree/hat/product
cat: /proc/device-tree/hat/product: No such file or directory
pi@rpi-stretch-full:~ $ 
```
As expected: HAT not recognized (0x68 instead of 'UU' and no '/proc/device-tree/hat/product').

Finally with PiJuice HAT on 0x52 and Sense Hat:
After boot the Sense HAT starts with a rainbow display on its LEDs, but the
LEDs go all off after a few seconds.
```
pi@rpi-stretch-full:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- 14 -- -- -- -- -- -- -- 1c -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- UU -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- 5c -- -- 5f 
60: -- -- -- -- -- -- -- -- 68 -- 6a -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
pi@rpi-stretch-full:~ $ cat /proc/device-tree/hat/product
Sense HATpi@rpi-stretch-full:~ $ 
```
The 'UU' on 0x47 is the kernel driver driving the LED framebuffer.

So. It does work for me, but write unprotecting the ID EEprom is necessary to actually change its address (@ChristopherRush soemthing which is not documented yet)

sensemonster, 2018-05-21:
Right then - I just finished testing the Sense Hat on another Pi using the same SD Card - all good.
I'm going to reassemble the Sense Path with the original Pi and PiJuice then then carry out the procedure @tvoverbeek sets out ...... will report back asap.... the dang pogo pin keeps falling out at the slightest jiggle :-)

tvoverbeek, 2018-05-21:
@sensemonster Do not forget to update the PiJuice packages to 1.3.3 (normal apt-get update/upgrade procedure) and if needed update the firmware to 1.2 using the gui or cli.

sensemonster, 2018-05-21:
Yep, I can confirm that all is now OK:

`i2cdetect -y 1` now shows the attached snippet.

But we always have to have an "unknown" :-)  I didn't get a chance to check the 'ID EEPROM Write unprotect'.  Before I disassembled - including battery removal - to attach the Sense Hat on the spare Pi I had changed the EEPROM to 52.  As soon as I reassembled everything and applied electrons the whole thing lit up :-)  and the tests now work.

Nevertheless @tvoverbeek as far as I am concerned "you are the man"  !  

Anyway, yes, I had already applied the latest v1.2 firmware and apt update/upgrade.

So, many thanks for all of you for the attentive response.  Soon, I will be receiving periodic environment measurements from my polytunnel!  if I get the periodic on/off python script right :-)
[image]

ChristopherRush, 2018-05-21:
@tvoverbeek I will get this documented, thanks for highlighting it.

sensemonster, 2018-05-21:
Guys I am closing the issue.  Thanks again.

closed 2018-05-21 by sensemonster