PiSupply/PiJuice #128: Software:Readme.md:Wakeup Alarm GUI documentation

sensemonster, 2018-05-22:
Please can we have a little more clarity on the functionality offered through this GUI.  I am aware that one can script such functionality.  However it would be more useful to show an example setting for .e.g tor power on/off every day for 30 seconds.  I can see that there is a "Minutes period" but it appears that this is the shortest duration one can configure every day, every hour at every n seconds after DAY:HH:MM

The other thing that needs clarification is how the check-box "Wakeup enabled" relates to the two buttons "Set Alarm" and "Apply"

[image]

ChristopherRush, 2018-05-22:
*Set Alarm* will save the above settings to the Alarm variable. *Wakeup
enabled* will ultimately enable or disable the wakeup alarm function. Any
changes made to the above two options must be saved, therefore you must
*Apply* all settings to the PiJuice.

To set the Pi to wakeup every 30 seconds or which ever, every day, you will
need to check *Every Day* then add 30 to the *Second* text field. All
remaining text fields must be set to Zero.

We are working on documentation that will give real world examples for all
aspects of the PiJuice Software.

tvoverbeek, 2018-05-22:
Some additional info on Wakeup enabled.
When the programmed alarm goes off and 'Wakeup enabled' is off the RPi will *not* be booted when it is off at the time the alarm goes off.
When 'Wakeup enabled' is on when the alarm goes off the RPi will be booted (similar to the SW1 short button press).
However after the boot the 'Wakeup enabled' will be disabled. When you programmed a recurring alarm you will have to programmatically enable the 'Wakeup enabled' again at RPi boot.
See also the example in my comment https://github.com/PiSupply/PiJuice/issues/91#issuecomment-381324352

sensemonster, 2018-05-22:
OK so, chaps, the "Minutes period" is still unclear :-)

@ChristopherRush thanks for explaining but I think if we look at what is being offered by this GUI from purely a functional point of you it still appears confusing - to me anyway :-).
The wording "Set Alarm" and "Wakeup enabled" are almost the same thing - functionally speaking: for those of us who as yet don't know what you are doing "under the covers" to deliver the function we want.  That function is to have periodic "waking up".  So, to help this dumbo anyway :-),  we need to establish why it is *not OK* to click the button "Set Alarm" and expect the functionality to be implemented.

@tvoverbeek thanks again for explaining.  However I am now thinking that after boot, when "Wakeup enabled" is disabled, I have lost the recurring functionality I was seeking :-)

tvoverbeek, 2018-05-22:
@sensemonster I guess you need to understand something 'under the covers'. The alarm is implemented by the Real Rime Clock (part of the PiJuice microcontroller). When the alarm goes off the microcontroller has to wake up the RPi.
The  recurring alarm will continue in the microcontroller, but you only have to enable the wakeup again at boot (i.e. in /etc/rc.local).
See my referenced comment above for a working example of a recurring alarm.

sensemonster, 2018-05-22:
Hi @tvoverbeek, indeed I will be using your example as a model for what I'm implementing.  I understand that by scripting it the "Wakeup enabled" can be set per boot-up.
To be absolutely clear, if I were to just use the GUI to set up a Wakeup, once the MCU based Alarm is triggered and the RPi is "woken"  that's it!  No more recurring Wakeup possible because the "Wakeup enabled" is disabled

tvoverbeek, 2018-05-22:
@sensemonster Correct.

sensemonster, 2018-05-22:
Many thanks - again - to both.

closed 2018-05-22 by sensemonster