PiSupply/PiJuice #131: Sample pijuice_config.JSON

jdotbdot, 2018-05-23:
sibly a dumb question because I'm not very familiar with JSON.

Is there any reason why in the sample config file at 
https://github.com/PiSupply/PiJuice/blob/master/Software/README.md,
under "user_functions", USER_FUNC8 is listed before USER_FUNC1?

Does the line `"enabled":true,` between `"min_charge":{...}`, and `"wakeup_on_charge":{...}` indicate some missing function(s)?

If I edit this file by hand, how do I get the PiJuice to recognise the changes?

tvoverbeek, 2018-05-23:
@jdotbdot If you do not want to reboot then you have to send the SIGHUP signal to the pijuice service (pijuice_sys.py).
Example in a bash shell:
Find the process id (pid) of the pijuice_sys.py process and send the signal:
```
pi@rpi-stretch-full:~ $ ps ax | grep pijuice_sys | grep -v grep
  372 ?        Ss     0:12 /usr/bin/python /usr/bin/pijuice_sys.py
pi@rpi-stretch-full:~ $ sudo kill -SIGHUP 372
pi@rpi-stretch-full:~ $ 
```
Note, you need `sudo` because the pijuice service is started under the root user.
When pijuice_sys.py receives the signal, it will reload the json file.

shawaj, 2018-05-23:
@ChristopherRush would you mind documenting this somewhere?

ChristopherRush, 2018-05-24:
Do we know the answer to the below query? @tvoverbeek @shawaj 

Does the line `"enabled":true,` between `"min_charge":{...},` and `"wakeup_on_charge":{...}` indicate some missing function(s)?

ChristopherRush, 2018-05-24:
I would assume it has something to do with system task enabled?

tvoverbeek, 2018-05-25:
Yes, **sudo systemctl restart pijuice** works too.
But when sending the SIGHUP signal to pijuice_sys the service is **not** restarted, but the signal is caught and treated directly by pijuice_sys.
Similar to many unix/linux utilities to force them to reread their configuration file.

ChristopherRush, 2018-05-25:
Documented

closed 2018-05-25 by ChristopherRush