PiSupply/PiJuice #133: Graphical interface too large on little screens

sakura-lain, 2018-05-25:
Hi again,

The graphical interface used to manage the PiJuice is too large on my 5" touch screen, it doesn't automatically adapt to the size of the screen and it seems impossible in that case to scroll down or reduce the window. Is there something I missed ?

Thanks again.

ChristopherRush, 2018-05-25:
Minimum window size is 400x400, try using the pijuice_cli.py from the terminal.

shawaj, 2018-05-25:
What's the resolution of the display?

tvoverbeek, 2018-05-25:
@ChristopherRush Note the size of the HAT configuration window is 640x480

ChristopherRush, 2018-05-25:
@tvoverbeek Yes I had seen that in the code and also `root.minsize(400, 400)`

ChristopherRush, 2018-05-25:
@shawaj 
[image]

tvoverbeek, 2018-05-25:
Just checked on a small terminal. Thanks to python-urwid it works fine.
If there are more entries than fit on the screen you get a scroll bar and you can scroll up/down with the arrow keys.
Also when mouse is supported in terminal, mouse works too.

tvoverbeek, 2019-02-05:
From https://www.raspberrypi.org/documentation/hardware/display/troubleshooting.md:
```
Some windows are cut off at the bottom of the screen so I can't use them

If some windows in X are cut off at the side/bottom of the screen, this is unfortunately a side-effect of developers assuming a minimum screen resolution of 1024x768 pixels.

You can usually reveal hidden buttons and fields by;

right clicking on the edge or top of the window,
picking "move"
using the up arrow key to nudge the window up off the top of the screen
If you don't have a mouse, see the right click fix below.
```
Does that help?

ChristopherRush, 2019-09-15:
Solution is to use the CLI like this - https://github.com/PiSupply/PiJuice/issues/133#issuecomment-392089245 or refer to this comment - https://github.com/PiSupply/PiJuice/issues/133#issuecomment-460639369

closed 2019-09-15 by ChristopherRush