PiSupply/PiJuice #136: PiJuice provides a mechanism to run arbitrary code as root.

jdotbdot, 2018-05-26:
In thread https://github.com/PiSupply/PiJuice/issues/31, tvoverbeek says of the config file and user scripts:

> Normally the pijuice_config.JSON file is owned by user pi, group pi since the python gui app is run as user pi when started from the Desktop.
> The directory **/var/lib/pijuice should have protection rwxrwxrwx** otherwise the python gui cannot write/update the pijuice_config.JSON file.

and

> Since **your script will be executed by the pijuice service it runs as user root**, not pi.

Note that on my Pi /var/lib/pijuice/pijuice_config.JSON belongs to root not Pi. **However, it has permissions rwxrwxrwx**. 

This is a huge security risk!

Anyone who can run the PiJuice GUI could make a button press run malicious or foolish code. 
Anyone who can access the command line can edit /var/lib/pijuice/pijuice_config.JSON so that a button press runs malicious or foolish code.
Can a system event such as a battery level be made to run a user script too?

Config files should be owned by root and be accessible only by root. 
User scripts should not be executed as root.

tvoverbeek, 2018-05-26:
Agreed on the security risk.
Since pijuice_gui needs to be able to update the json file it needs access to it.

What can be done is to run any user script started by the pijuice service under the userid of the owner of the user script.
Still does not help very much if the owner is 'pi' since user 'pi' can run any command as root because 'sudo' does not require a password for 'pi'

shawaj, 2018-05-26:
Any updates to improve security are definitely a good thing. What would be your recommendation @tvoverbeek @wikipedia555 ?

jdotbdot, 2018-05-26:
> Since pijuice_gui needs to be able to update the json file it needs access to it.

Hmm the GUI can make changes to /etc/wpa_supplicant/wpa_supplicant.conf too. And that file belongs to root and has permissions 600

jdotbdot, 2018-05-26:
One way to allow a low privileged user to modify protected files is for the (compiled) program which actually writes the file to belong to root and have the setuid bit set. That's an old technique though, modern practise may be better.

jdotbdot, 2018-05-27:
I think the essential change is for scripts to run as the script owner. 
I'm likely to write a script which includes "sudo shutdown -h now". That should be no more likely to succeed via PiJuice than it is from the command line - ie the user must be a sudoer.

Clearly PiJuice itself needs to be able to issue a shutdown instruction. Does that mean it has to run as root?

change /var/lib/pijuice and contents to belong to pijuice:pijuice and have permissions 755
Now I have to use sudo to edit it.

tvoverbeek, 2018-05-28:
@jdotbdot @wikipedia555 @shawaj 
Found a solution which is tested and working.

@jdotbdot Yes, pijuice_sys needs to be able to issue the 'sudo halt' instruction. Even when running as user pijuice this can be done by adding an appropriate sudoers definition for user pijuice executing /sbin/halt (See below).

With permisions 710 on /var/lib/pijuice (owned by pijuice) you can still edit the config file as user 'pi' (or as any user belonging to the 'users' group) by specifying the complete path (/var/lib/pijuice/pijuice_config.JSON) to your favorite editor where pijuice_config.JSON has permissions 660.
Note the directory permission has changed from yesterday (-wx--x--- to rwx--x---)
This works because user pijuice belongs to the 'users' group (gid 100) as does the default user 'pi'.
If you introduce an other user you will have to manually add it to the 'users' group.

Here is what else needs to be changed:
The pid file created by pijuice_sys (now /var/run/pijuice.pid) has to be moved to /var/lib/pijuice/pijuice.pid since /var/run is owned by root and normal users do not have write access.
Of course the pid file defintion in pijuice_gui and pijuice_cli has to be adapted accordingly.

Add a sudoers file `020_pijuice-nopaswd` to the /etc/sudoers.d directory with the following contents:
```
pijuice ALL=(ALL) NOPASSWD: /sbin/halt
pijuice ALL=(%users) NOPASSWD: ALL
```
This allows the pijuice user to execute the halt command and to execute any program/script when running as a user belonging to the 'users' group.
The latter is used to execute the user_functions as the owner of the function/script.
In pijuice_sys first the owner of the script is found, then the script/function is called with
`sudo -u owner function arguments`

I can prepare a PR with these changes if you (@shawaj @wikipedia555 @PiSupply/software) agree

jdotbdot, 2018-05-28:
Yes, I note that you are asking the product team not me, but just pointing out...

Moving a pid file from a file system stored in memory ( /var/run is a link to /run; /run is a tempfs ) to a real file system on the sd card ( /var/lib is on the root filesystem ) might result in orphan files if the Pi halts unexpectedly.

> you can still edit the config file as user 'pi' (or as any user belonging to the 'users' group

IMHO the config file should not be editable without sudo.

tvoverbeek, 2018-06-01:
@jdotbdot OK, Did some more thinking and fiddling.

The service (pijuice_sys.py) is started as user pijuice.
pijuice_cli and pijuice_gui run as user pijuice using a C-coded setuid wrapper.
Execute permission removed from pijuice_cli.py and pijuice_gui.py

User pijuice primary group is now pijuice.
/var/lib/pijuice is only accessible for owner pijuice, group and other permission bits are 0.
Result: users can only edit pijuice_config.JSON with sudo.

Before a user script (or program) is started by the service it is checked the script/programs owner is not root, has execute permission and belongs to group pijuice. Only then the script/program is started with 'sudo -u <owner>'
Implies the owner of the script/program will have to be added to the pijuice group.

Some other changes are needed because of running as non privileged user pijuice (sending signals to the service and the tray app, pid files moved to /tmp)
Will document this with more details

@shawaj @PiSupply/software @wikipedia555 What do you think? I believe this increases security quite a bit.

ChristopherRush, 2019-09-15:
Fixed here - fcf81d819e2189eaa53fcec56817027d4cfbc7fa

closed 2019-09-15 by ChristopherRush

Ruben138, 2025-02-24:
Yo