PiSupply/PiJuice #139: dumb question about wakeup :)

MattsProjects, 2018-05-28:
I probably just missed it in the documentation I think...
Could you tell me which type of wakeup takes precedence, wake on charge or wake alarm?
At the moment I have both enabled (wake every hour on the 0 minute, and wake at 20% charge. Pin is installed).
But I'm not sure if the wake alarm will keep happening if the battery is below 20%.
I also have the minimum charge set to 5% (it will halt and power off at this level).

My concern is that if I get down below 5%, I might have a situation where the rpi wakes up and is immediately shut down in the middle of boot up, risking corruption.

However, if the wake-on-charge takes priority and the alarm is ignored until 20% is reached, then all is well :)

Thanks!!

MattsProjects, 2018-05-28:
PS: if what I am doing puts me into a no-mans-land, I'm happy to pick one wakeup strategy and work around it manually using the API and some scripts :)

cmsimike, 2018-06-01:
Sorry to jump into this, but you said "wake on charge?" Is there a setting somewhere that will power on the pi once the pijuice detects it is being charged again?

I feel like I went through the GUI to find this, but wasn't able to. Where did you find it?

tvoverbeek, 2018-06-01:
Setting is on the 'System Task' tab of 'PiJuice Settings'

MattsProjects, 2018-06-01:
Yep, sorry for the ambiguity. Pijuice will wake the pi when the battery has charged above a certain percentage.
FYI, my use case is this:
1. Run Pi as normal, with a USB source (basically it is just my bedside weather display box).
2. If USB power is lost, send me an email, and start some 'standby protocol'
3. The standby protocol is:
A. shut down pi after 60 seconds (pijuice then shuts down 60 seconds later)
B. wake up every hour and check for USB power.
C. repeat A & B until USB power is reconnected.

So the question is, what happens if USB power never comes back?
Eventually the battery dies of course, but it has to go through a gray area defined by the minimum charge and/or minimum voltage parameters in the PiJuice.
So I was just curious if the wake alarm takes precedence over the wake on charge.
If it's ambiguous, then I just add in some code to behave as I like in the gray zone.
(ie: if at step B, the battery is at 4%, just shut down and cancel the scheduled wake up.)

tvoverbeek, 2018-06-01:
'wake alarm' and 'wake on charge' have equal priority.
Whoever comes first wins (provided the PiJuice MCU remains running)
If PiJuice goes off because battery is empty 'wakeup on charge' is reset at PiJuice power on.
So I guess your best bet is to add your own code to deal with low battery.
Set wakeup on charge high enough, because as soon as you load the battery (RPi booting) the
reported charge level drops significantly.
See also the "Battery charge level notes" in the hardware README on github.

closed 2019-09-15 by ChristopherRush