PiSupply/PiJuice #141: PiJuice won't shut down

rbnpi, 2018-05-29:
Installed latest 1.3.3 package from raspbian. Upgraded firmware using this as prompted to 1.2_2018_05_02.wlf.binary
SW1 doesn't shutdown, or turnoff Pi no matter how long I push it.
If I do a shutdown from the Pi, then SW1 will restart it.
However, to completly shutdown, since SW1 won' remove power with a 20s push I have to remove the battery after the Pi has stopped all activity.

Installed pogostick pin, and saet tab to pin installed but I can't get wakeup to work.

rbnpi, 2018-05-29:
When I powered down I had no external connection to the Pi. It was running off the pi_juice battery (I started it with power applied to the Pi, which was then removed). The settings in the gui for SW1 are
single press HARD_FUNC_POWER_ON 800
Long press 1 SYS_FUNC_HALT 4000
Long press 2 HARD_FUNC_POWER_OFF 8000

tvoverbeek, 2018-06-01:
OK. Closing this one.

closed 2018-06-01 by tvoverbeek