PiSupply/PiJuice #142: PiJuice Status LED's not working.

w4qed, 2018-05-30:
None of the three status LED's are working on my PiJuice HAT I am sure it is probably something I've done. I have tried to reset everything to default, but still nothing. The power on  and shutdown seem to work correctly so the switches are still defined OK. Any assistance would be appreciated.

tvoverbeek, 2018-06-01:
@w4qed Since the switches are working, the first thing to do is to check the LED status on the LEDs tab of the 'PiJuice HAT Configuration' (assuming you use the settings gui).
Default settings are for D1: CHARGE_STATUS, R 60, G 60, B 100 and for D2: USER_LED, R 0, G 0, B 0

ilyastrodubtsev, 2018-06-01:
@w4qed I can't replicate this on last firmware. Can you, please, add a steps to replicate.

w4qed, 2018-06-01:
@tvoverbeek Thanks for the response. I did check and in one case I changed.
D1 was OK at R 60, G 60, B 100, so nothing changed there.
D2 was R 10, G 25, B 100 so changed to R 0, G 0, B 0.

 Well that did not change anything, Still no LED's lit.

w4qed, 2018-06-01:
@ilyastrodubtsev Well not sure how to replicate. LED's are still OFF or not lit.

w4qed, 2018-06-01:
Sorry messed up and did not understand the meaning of comment buttons, so...

 None of the three status LED's are working on my PiJuice HAT I am sure it is probably something I've done. I have tried to reset everything to default, but still nothing. The power on and shutdown seem to work correctly so the switches are still defined OK. Any assistance would be appreciated.

ChristopherRush, 2018-06-01:
@w4qed Have you tried a hard reset by removing the PiJuice battery and external power?

ilyastrodubtsev, 2018-06-01:
@w4qed  What version of the firmware do you have?

w4qed, 2018-06-01:
@ChristopherRush  Yes I have tried, left the PiJuice Hat unplugged and battery removed for abut 6 hours Assembled and same no LED's.

w4qed, 2018-06-01:
@ilyastrodubtsev Thanks, I'll try attaching photo, but firmware states 1.2 no updates.
[image]

tvoverbeek, 2018-06-01:
@w4qed What functions are indicated for D1 and D2 on the LEDs panel?
With D2 (assuming it is on USER_LED) and RGB = 0,0,0 it will be off.
If D1 happens to be on 'NOT_USED' it will also be off irrespective of RGB settings.
With no power connected, Pi off and PiJuice off (but battery has charge) do you have a weak green blink on D1 every 4 seconds?
If not, then my guess is you either have faulty LEDs or a corrupted firmware problem.

w4qed, 2018-06-01:
@tvoverbeek  here is my setup. Doing pictures so not to misrepresent my actions.
[image]

tvoverbeek, 2018-06-01:
@w4qed Settings are OK
What about Pi off and PiJuice off (in fact in low power mode)?

w4qed, 2018-06-01:
@tvoverbeek Not sure what Pi off and PiJuice off? When Pi is shutdown, no LED's on Hat or Pi when shutdown with or without power connected. I've only been using the PiJuice Hat power in jack.

tvoverbeek, 2018-06-01:
@w4qed Then it seems like a hardware or corrupted firmware problem.
Have you tried to reload the firmware manually?
Command is 'pijuiceboot 14 /usr/share/pijuice/data/firmware/PiJuice-V1.2_2018_05_02.elf.binary' in a terminal window.
Look at the output from pijuiceboot.
When upload is successful the firmware should restart and the D2 led should be bright blue.

w4qed, 2018-06-01:
@tvoverbeek Well did the command and seem to do a 256 write of data and completed OK. !st reboot pi, no change. The shutdown Pi, no change shutdown Pi removed Hat and battery, waited 3 minutes and put everything back and booted it; still no change. Everything seems to work but the LED's. I do want to thank you and all the others for your assistance.

w4qed, 2018-06-01:
@tvoverbeek  I did notice one difference in that when Pi is shutdown and power removed from the PiJuice that the RED Led on Pi is on.

tvoverbeek, 2018-06-01:
@w4qed When the red led on the Pi is on, the PiJuice is still supplying power to the Pi and not in low power mode. A short press on SW1 should then boot the Pi.
Try pressing SW1 for about 20 seconds (with default SW1 settings) then the PiJuice should remove power from the Pi -> red light on Pi off.
With no USB power connected the PiJuice should then be in low power mode.

w4qed, 2018-06-01:
@tvoverbeek Yes that did turn off the Pi's LED.

w4qed, 2018-06-01:
Is there a PiJuice Schematic? Probably not as China would probably rip them off; then again may be made in China in that case, oops too late.Looks like Canada perhaps?

w4qed, 2018-06-01:
Found the STM32F030CCT6.pdf Just not sure what drives the LED's.

closed 2019-09-15 by ChristopherRush