PiSupply/PiJuice #143: PiJuice RTC incorrect time

ChristopherRush, 2018-06-01:
Posting on behalf of a user:

"Any idea why the time in Wakeup Alarm is set to UTC time? My Pi clock is set to Zurich time, and I cant seem to Sync them"

tvoverbeek, 2018-06-01:
@ChristopherRush Linux always keeps its internal clock on UTC. When presenting time to the user it uses the timezone info to convert it to local time. Therefore it can do things like standard/summer time changes automatically.
The RTC is used as a backup for the Linux internal clock when Linux is down. Hence it uses RTC.
Otherwise you would have to change RTC settings every time you change timezone or at summer/standard time changes.
Since the alarms are provided by the RTC they have to be specified in UTC.

aixsyd, 2018-06-01:
Good to know, wanted to make sure I wasn't missing anything.

ChristopherRush, 2018-06-04:
@tvoverbeek Thanks, will document this and close.

rossirm, 2018-07-30:
Hi, there!
Can anyone describe me step by step (as to a dummy) how to make Raspberry set its time from the RTC of my PiJuice independently of an Internet connection. I use my Raspberry with PiJuice and SenseHat, which wanted to change the address of (EEPROM or I2C) to 0×52 from 0×50 in PiJuice software.
Thanks in advance!

closed 2019-09-15 by ChristopherRush