PiSupply/PiJuice #144: Ticking noise?

github-0, 2018-06-01:
Just noticed my PiJuice is making a ticking noise when it is powering the Pi. It's 
- faint, I have to bring it almost next to my ear to hear it 
- fast, about 8 times per second
- when powering off using SYS_FUNC_HALT_POW_OFF and the Pi has halted but not yet fully powered down by PiJuice, the "ticks" are matching PiJuices power LED (=slow ticking instead of fast)

Normal or not? I hadn't noticed it before installing the pogo pin today.

ChristopherRush, 2018-06-01:
Yes, I have noticed this also. No doubt it is some power component doing some switching.

aixsyd, 2018-06-01:
I have not noticed the sound, but I have noticed that when I am powering the Pi via Pijuice and I set a wakeup alarm and do 

`sudo shutdown now`

the Pi closes my SSH session, but I can still ping the Pi and it never actually shuts down. What's that about?

shawaj, 2018-06-01:
Excuse my childishness but this really reminded me of this - https://youtu.be/Tx1XIm6q4r4

Having said that, as Chris says I wouldn't worry about it.

However will check with @mmilann anyway

github-0, 2018-06-01:
Excused :)

And thank you, I thought it could be normal but tbh it seemed very unlikely that I wouldn't have noticed it before that I got a little paranoid. Waiting for the official "all clear" just to be safe :D

@aixsyd I think that's normal behavior of the Pi and Linux in general, you need to use -h to actually shut down (or halt) the system (even then I'm not sure if the Pi still responds to ping). This topic/issue is not about shutting down though :)

ChristopherRush, 2018-06-04:
@aixsyd Technically running `sudo shutdown now` does not shutdown the Pi as we know it. It enters some sort of recovery mode that can only be accessed via the physical device. You need to use the option `-h` to fully shutdown and halt the processes and CPU. `sudo shutdown -h now`

tvoverbeek, 2018-06-04:
@aixsyd @ChristopherRush The 'some sort of recovery mode' is known as single user mode.
Normally used by root for system maintenance mode with no other users logged in.
Network will still be up, this is why uou could still ping it.
However logins are disabled (-> no remote ssh access)

github-0, 2018-06-09:
@shawaj @mmilann Any news on this? Is my Pijuice going to blow up? Would be interesting to know which part is causing this.

ChristopherRush, 2018-07-03:
@mmilann Any update as to what might be causing this?

shawaj, 2018-07-05:
Pinged Milan by email for an update on this. Sorry for delay @github-0

shawaj, 2018-07-05:
From @mmilann :

Hi Aaron,

If everything works and factory tested ok then no problem, inductor involved in DC/DC converter circuits usually make noise, and how much depends on production variation, some may appear to have audible noise

closed 2018-07-05 by shawaj

hcfman, 2023-07-09:
Yes, but I bought 3x of these to use in an audio recording project. The ticking goes into the recording. This is a show stopper.

There should be a solution to this problem or it should be make evidence in the specifications.

hcfman, 2023-07-09:
It's not just audible, it's also a spike on the power rail. A USB mic place far away from the pi juice hat still ends up with it in the audio recordings. I'll likely have to return my units and find another one without this problem, considering that this has been noted 5 years ago and people still have the problem.