PiSupply/PiJuice #146: USERFUNCx changes on Buttons tab do not get picked up by the pijuice service

tvoverbeek, 2018-06-03:
Changes on the Buttons tab of ‘PiJuice HAT Configuration’ are updated
in the PiJuice MCU when you press the ‘Apply’ button.
However if you change e.g. USERFUNC6 to USERFUNC7 for a button press,
this change does not get picked up by the pijuice service.
It will still execute USERFUNC6 after the change.
This is because the USERFUNCx assignments are only picked up when the
service starts.

Easy fix: Include the USERFUNCx assignments in reload_settings() of
pijuice_sys.py and add a notify_service() call in pijuice_gui.py when
you press the ‘Apply’ button

Fix already committed (https://github.com/PiSupply/PiJuice/commit/c0d6f6048333cf7589460dabd8b127c74e066ffe)
Raised issue to document it.
This issue can be closed after the next version is released.

sensemonster, 2018-06-04:
May I ask if this also applies to the case when setting up a USERFUNCx function for the first time?

I set up USERFUNC1 to run a python script off SW3 'PRESS'.  The python script simply disables a specific cron job.  I haven't been able to get it working so far :-)

tvoverbeek, 2018-06-04:
@sensemonster Yes it does.
To get the service to recognize your change in the button functions in the released version you have to restart the service either by rebooting or (easier) the command `sudo systemctl restart pijuice.service`

sensemonster, 2018-06-04:
OK I will give that a try.  Thanks again

closed 2019-09-15 by ChristopherRush