PiSupply/PiJuice #147: Some examples of simple scripts using PiJuicePower management functions

richardquint, 2018-06-06:
As a total newbie with python and the PiJuice hat I'd appreciate some examples of simple scripts---e.g., a script to execute sudo halt followed by a SYS_FUNC_HALT_POW_OFF.  Using the PIXEL desktop I select MENU|Shutdown|Shutdown followed by a button press assigned to SYS_FUNC_HALT_POW_OFF.
There must be a more elegant and efficient way to do this.

TIA

tvoverbeek, 2018-06-07:
No need to go via the PIXEL desktop route to shutdown the Pi.
The button press alone (if assigned to SYS_FUNC_HALT_POW_OFF) is sufficient.
After holding SW1 pressed for 10 seconds (with default duration setting for LONG PRESS1) you see LED D2 flickering, meaning the PiJuice sent the shutdown command to the Pi.
Condition is that the System Task is enabled on the PI, otherwise it does not work.
Then 60 seconds after the sending of the command the PiJuice removes power to the Pi (Red LED on Pi goes off), provided you power the PiJuice via its own USB.
Obviously if you power the Pi via its USB, there is no way the PiJuice can remove the power to the Pi.

Here is a Python script (myshutdown.py) which does the same thing:
```
#!/usr/bin/env python

import os
from pijuice import PiJuice

pj = PiJuice(1, 0x14)

# Remove Pi power after 30 seconds
pj.power.SetPowerOff(30)

# Shutdown
os.system("sudo halt")
```
If you make the script executable (`chmod +x myshutdown.py`)
you can start it with `<path to>myshutdown.py`

richardquint, 2018-06-07:
Perfect.  Just what I was looking for.  Thank you.

ilyastrodubtsev, 2018-06-18:
I consider the problem closed. If the user has new questions, it will be necessary to create a new issue.

closed 2018-06-18 by ilyastrodubtsev