PiSupply/PiJuice #149: Ability to use software i2c on GPIOs

Leobaillard, 2018-06-09:
Hi!

First of all, thanks and congratulations for an amazing product!

I'm trying to create a pico-projector project using the DLP 2000 Evaluation Module from Texas Instruments. I'd like to run it on battery using a PiJuice. The problem is that it uses DPI pins which require to move the i2c pins on other GPIO ports. Here is the dtoverlay for that: 
`dtoverlay=i2c-gpio,i2c_gpio_sda=23,i2c_gpio_scl=24,i2c_gpio_delay_us=2`

I was wondering if it would be possible to still use the PiJuice in this case? I understand that I might not be able to plug it directly onto the Pi regarding the pin assignments on the physical header, but maybe using cables?

When running like this, it appears to use the i2c bus 3 as there is a `/dev/i2c-3` available. I tried to change the `BUS` variable in `pijuice_cli.py` but even though the program opened, it could not see the PiJuice and crashed when trying to access a page. By the way, what pins other that +5V, GND and I2C are required to be connected for the PiJuice to work?

Thanks for the answers that you can provide!

tvoverbeek, 2018-06-09:
I guess this is similar to http://frederickvandenbosch.be/?p=2948

You have to change to i2c-bus 3 in one more place in addition to pijuice_cli.py:
The system service `/usr/bin/pijuice_sys.py`:
In the main() function:
```
try:
        pijuice = PiJuice(3, 0x14)   <=====
    except:
        sys.exit(0)
```
If you intent to use the Desktop on your pico-projector
also `/usr/bin/pijuice_tray.py` and `/usr/bin/pijuice_gui.py` will need the same update.

You will have to use jumper wires to connect the PiJuice (as Frederick did to connect the DLP2000).
As far as I know the only GPIO connections needed for the PiJuice are I2C, GND and 5V.

Hope you get everything stacked and working properly.

Leobaillard, 2018-06-11:
Thanks! This is exactly the article I was basing my project on.

It seems simple enough, I will try that and report here if all is working as expected. It could be interesting to others that want to try similar projects and use a PiJuice with it!

Thanks again!

tvoverbeek, 2018-06-11:
I do not believe you can drive both the DLP2000 and the RPi with a PiJuice.
The DLP requires a 5V-3A power supply.
3A is far above what the PiJuice can supply.
You need an extneral supply for the DLP

Leobaillard, 2018-06-11:
I'm not sure that's the case: the power supply sometimes shipped with the DLP 2000 is indeed 5V-3A, but is largely overkill. MickMake also created a picoprojector project using this board and a Raspberry Pi Zero W. He measured the power consumption of the devices when operating and found that the total, under load, was under 700mA.

I get that the Rpi 3B+ I'm using consumes more power than a Rpi Zero W, but it might be low enough to allow use with the Pijuce, right? What is the max power output of the PiJuice? Maybe this can only be achieved with a Pi Zero. I'll have to try :)

Source: https://www.mickmake.com/post/build-a-pi-zero-w-pocket-projector-project/5

tvoverbeek, 2018-06-11:
From the hardware README on github:
```
Max. current that the PiJuice will be able to supply
The maximum current at 5V via the GPIO is 2.5A and via VSYS is 2.1A. This is also dependent on
battery capacity. For BP7X battery for example we have measured around 1.1A at 5V GPIO and around
1.6A at VSYS output. To achieve the maximum of 2.5A you will need to use a battery of at least
3500mAh.
```
@shawaj How does this match your 3A?

Anyway 700mA should work

shawaj, 2018-06-11:
@tvoverbeek must have misremembered! My bad. Not sure where I got the 3A from?!?

FYI @Leobaillard we should have a Pi hat version of that projector soon. Hopefully shipping by the end of the year. Just finished the design stage an on to prototyping.

Leobaillard, 2018-06-11:
@shawaj really?! That's awesome! What can I do to get updates about that project? Thanks again!

Leobaillard, 2018-06-11:
I did some testing. Unfortunately, I could not make it work with my current configuration (DLP 2000, Rpi 3B+ and PiJuice with standard BP7X battery).

I fully charged the PiJuice battery, then connected pin 2 (+5V) and 6 (GND) of the PiJuice directly to the Pi (on the same pins) using jumper cables and pin 3 (SDA) and 5 (SCL) of the PiJuice header to the P2 header of the DLP 2000 EVM on pins 20 (SDA) and 19 (SCL). I then changed the Python files to connect to I2C bus 3 and booted the Pi.

The two LEDs on the projector board lit up one after the other and cycled a few times before it gave up trying to boot the board. Sometimes the lower LED would stay on, sometimes not. The lamp however, never turned on and the Pi would run without any issue (though only the green activity LED would lit, not the red one).

The other problem I encountered was with the communication between the Pi and the PiJuice. It kind of flickers: I sometimes get data, and sometimes get errors. Mostly errors though. The GUI appears and sometimes the battery tray icon shows the battery percentage, but I get a lot of communication errors or incomplete data.

I knew it was a long shot, but hey, thought I'd test it anyway. I understand why the projector might not be turning on: there's probably not enough current available ; but I'm not sure about the I2C bus getting scrambled data. If you have any ideas regarding that, I'm all ears :)

Maybe a bigger battery would do the trick as you mentioned that higher current could be availableusing one of those? Or a regular Pi 3B or even a Pi Zero?

tvoverbeek, 2018-06-12:
The red LED on the Pi not coming on means it detected undervoltage (lightning bolt in the upper right hand corner of the desktop).
```
Finally on the subject of icons, in the past if your Pi was working particularly hard, you might have
noticed some yellow and red squares appearing in the top-right corner of the screen, which were
indications of overtemperature or undervoltage. These have now been replaced with some new
symbols that make it a bit more obvious what’s actually happening; there’s a lightning bolt for
undervoltage, and a thermometer for overtemperature.
```

Leobaillard, 2018-06-30:
Ok, I tried on a Pi Zero W with the WiFi off, same result. The DLP projector does not initialize. I don't know how it assesses that the power is not sufficient, but it seems it's not.

I hooked up the Pi Zero to my multimeter to check the current draw, I'm at ~200mA when fully powered on. So if Mick Make's measurements are to be trusted, I should be under the 1.1A that the PiJuice can deliver with the standard battery, right?

I event tried plugging the DLP only to the GPIO ports on the PiJuice, without the Pi being connected. Same result. Is there some some of negotiation that needs to happen on order to get the full power that the PiJuice can deliver? Or do you think the issue could be elsewhere?

All that's left to try is to get a bigger battery I guess. Any advice on that part?

tvoverbeek, 2018-07-01:
@Leobaillard Did some experimenting with bit-banged I2C and the PiJuice using GPIOs 23 and 24 (pins 16 and 18 on the GPIO connector).
One important thing is you need **external pull-up resistors** when using GPIOs 23 and 24.
Without them I2C does not work. I understand the projector also uses I2C commands (address 0x1b), so in case you did not use pull-up resistors to the 3.3V rail this might explain why the projector did not work.
I used 2k2 for the 2 pull-up resistors. The PI's i2c-0 and i2c-1 have internal pull-up resistors of 1k8.
With the pull-up resistors in place the PiJuice software works as usual (of course modified to use I2C bus 3 instead of 1)

doronhorwitz, 2019-02-22:
I've tried to follow all the comments in this issue but have gotten a bit lost.
I too want to connect up a PiJuice to a Raspberry Pi 3 but am already using the I2C pins (BCM2 and BCM3).
From reading everything from all the comments above, I understand the following:

1) The `dtoverlay` must be modified to make I2C available on free GPIO pins
2) I need to modify `pijuice_cli.sys` and `pijuice_sys` to use `/dev/i2c-3`
3) I should use pull-up resistors to 3v3 on the GPIO pins which have been set for I2C - assuming they don't already have pull-up resistors
4) I'll have to use jumper cables to connect these new software GPIO pins to the I2C pins of the PiJuice since using the 40 pin header to connect the PiJuice to Raspberry Pi 3 will not work.

Am I correct about all of this?

If I am correct, is there really no way to change the pins the PiJuice uses for I2C? I really want to use the 40 pin header directly like HATs should, and avoid having to use jumper cables.

tvoverbeek, 2019-02-22:
@doronhorwitz You wrote:

> I too want to connect up a PiJuice to a Raspberry Pi 3 but am already using the I2C pins (BCM2 and BCM3)

Question: Are your using BCM2 and BCM3 to address an other I2C peripheral or do you use them for an other purpose?

doronlh, 2019-02-23:
@tvoverbeek, they are used for an I2C peripheral

tvoverbeek, 2019-02-23:
@doronhorwitz As long as the other I2C peripheral does not use the PiJuice I2C addresses (0x14 and 0x68) you can use them in parallel without any problems. No need to use bit-banged I2C on another pair of pins. I2C is a serial bus which supports multiple devices.

doronhorwitz, 2019-02-23:
Thanks for clarifying my confusion @tvoverbeek and improving my understanding of I2C.
This issue confused me because I saw someone talking about jumper cables above. But I understand now, that for that case it's because the other peripheral being used uses BCM2 and BCM3 for other purposes and not I2C.

closed 2019-09-15 by ChristopherRush