PiSupply/PiJuice #151: PiJuice based Pi Case with 3 x external buttons that contact SW1, 2, & 3

sensemonster, 2018-06-11:
May I ask if there are any plans to offer a case as above?

shawaj, 2018-06-11:
Have you seen the PiJuice tall and short cases?

sensemonster, 2018-06-11:
Yes, forgive me but have failed to discern the button access in the cases.  The pictures on coolcomponents or modmypi only describe "openings".   Exactly what does this mean please?  some holes for a screwdriver?

shawaj, 2018-06-11:
Yes exactly for a screwdriver or pen lid. But you could probably hook up something more permanent

sensemonster, 2018-06-11:
Hmm I get it now :-) so it's no different from a hole drilled in any other case.  But seriously, I am trying to engineer two external buttons for SW1 & SW3.  The external buttons will be attached to a project box containing the boards.  Maybe the best option is to work out what SW2 & SW3 do and implement the same functionality via GPIO direct (?).

closed 2019-09-15 by ChristopherRush