PiSupply/PiJuice #152: Pijuice current supply with normal battery

github-0, 2018-06-11:
Quick question:

_["For BP7X battery for example we have measured around 1.1A at 5V GPIO"](https://github.com/PiSupply/PiJuice/tree/master/Hardware#max-current-that-the-pijuice-will-be-able-to-supply)_

Have I understood correctly that if I power the Pi through Pijuices USB port or when the Pi is running on battery power alone, the maximum possible current available for the Pi with the default battery is 1.1A?

Considering the [power requirements](https://www.raspberrypi.org/help/faqs/#powerReqs), this doesn't necessarily leave much (or any) room for peripherals. How much does the Pijuice itself consume, does that count towards the 1.1A? How do we know if we are close/over the limit (other than unstable system), can Pijuice calculate current consumption?

ChristopherRush, 2018-07-04:
I connected two USB hard drives at the same time to the Pi and it couldn't boot any one of them up, they kept shutting down due to the lack of power. Plugging them in one after the other seemed to be ok in an idle state and it consumed at max 2A. I would imagine that the above statement to be true for a stable Raspberry Pi and will be suited for the majority of external peripherals.

ChristopherRush, 2019-09-15:
Alternative here is to use a much larger capacity battery, which we now stock in the Pi Supply store.

closed 2019-09-15 by ChristopherRush