PiSupply/PiJuice #154: Solar feedback

shawaj, 2018-06-14:
Received this from a customer:

The last point is one technical issue. There's a lot of talk on your site about solar power, but the PiJuice hat can't be powered from a standard solar panel. If there was anything I would change, I would increase the voltage regulator input to 15 volts. This way serious research with proper off grid portable solar panels could be done, without the need of additional noise making DC DC converters. This would allow me to plug the PiJuice hat directly into my 4S lithium iron phosphate battery pack which outpyts 12-14.4 volts.

shawaj, 2018-06-14:
Actually from the creator of this YouTube video https://youtu.be/BXYdsdqDkW4

klint-k, 2018-06-14:
Wow sounds familiar.

klint-k, 2018-06-27:
Would it be hard to get a sub hat like this run through all the hoops quickly?

PiJuice converter phat.
- 3 pogo pins
           One for the transfer from RPI3 "on jumper" to RPI3+ "on jumper"  for the trace.
           Two for a jumper to 8v -10v
- two voltage converters
            One from the PoE (24V to 36V) to 12.6V or PoE (24V to 36V) to 5V
            One from 11.8V to 15V to 9V
            Plus all the passive components
- two holders/standoffs for the pogos
- two different yet common barrel connectors
- 40 pin connector
- 4 pin connector  for the PoE