PiSupply/PiJuice #155: Battery on screw terminal

serge16, 2018-06-14:
In the hardware documentation it is written: "You can use any single cell lipo with PiJuice as far as you configure the board and the software correctly. **You cannot use cells/batteries in parallel or series.** "
Why is it not possible to connect on the screw terminal a battery holder with 2 or 3  18650 batteries mounted in parallel and protected by a PCM? The Pijuice should "see" it like one  battery, no ?

tvoverbeek, 2018-06-14:
Your PCM will be fighting with the built-in PCM of the PiJuice.
See the datasheet https://github.com/PiSupply/PiJuice/blob/master/Hardware/Datasheets/BQ24160RGET.pdf
of the chip used by the PiJuice which only supports single cell batteries.

fheslouin, 2018-06-20:
I was planning in using multiple cells LiPo battery, but I'm figuring out I can't. 
The file `PiJuice Battery Discharge Levels.xlsx` is talking about 5000 mAh and 12000 mAh where can we find them ?

klint-k, 2018-06-22:
How Quick could PiJuice make a converter board?

fheslouin, 2018-07-04:
Apparently there is no other option than the built in battery and an other 2300 mAh battery ?
What about the 12000mAh single cell option listed in the excel file ? 

It make this product less attractive than expected...

reddo10, 2018-07-04:
I cannot seem to get the PiJuice to behave when connecting a single cell 10000mAh battery to the screw connectors. When I set this up and introduce mains power to the PiJuice microusb, all starts ok. When I pull the mains power away, the battery keeps the Pi on by seamlessly switching to the battery power. When I reintroduce the mains power the Pi restarts? Why? I've tried altering the DIP switch to the lipo setting but this doesn't seem to alter anything. I've not done anything in the software settings. Has anybody had a similar experience, it better still had anyone had a successful experience they're prepared to share with me?

Justin

ChristopherRush, 2018-07-04:
@fheslouin You can connect any lithium polymer/ion single cell battery to the PiJuice via the provided screw terminal and configure the battery profile in the software. You can quite easily find these types of batteries from eBay (most likely a far east source) or Alibaba.

shawaj, 2018-07-05:
@reddo10 you already started another issue at #162 so to save confusion we will put all replies to you over there instead of here

ChristopherRush, 2019-09-15:
We now stock these larger batteries in the Pi Supply Store - https://uk.pi-supply.com/search?type=product&q=pijuice+battery

closed 2019-09-15 by ChristopherRush