PiSupply/PiJuice #156: Firmware Update

Ced0, 2018-06-14:
I believe, you did'nt indicate in the documentation that when you update the firmware, the power supply must be directly pluged into the raspberry, otherwise the juice is gonna get "broken".

tvoverbeek, 2018-06-14:
Huh?
I have updated the firmware many times (for testing purposes) with power connected to the
PiJuice USB input.
How does it get "broken"? Please some more details, so we can see if there really is a problem/bug.
Do you update using the GUI or do you use 'pijuiceboot' directly?

ilyastrodubtsev, 2018-06-18:
@Ced0 Good afternoon. Tell us how you update the firmware. Desirable in great detail.

ChristopherRush, 2018-06-18:
@ilyastrodubtsev It is documented in the following places 

https://github.com/PiSupply/PiJuice/tree/master/Software#pijuice-hat-config-firmware-menu

https://github.com/PiSupply/PiJuice/tree/master/Software#firmware

https://github.com/PiSupply/PiJuice/tree/master/Firmware

ilyastrodubtsev, 2018-06-18:
@ChristopherRush Thanks for the answer. I've updated the firmware many times already for tests. I guess the user could have made a mistake somewhere.

Ced0, 2018-06-18:
I couldn't reproduce the issue. I'm closing the issue. 
Thanks for your help.

closed 2018-06-18 by Ced0