PiSupply/PiJuice #157: Communication error with Raspberry Pi 7" Touchscreen Display (Official)

opbillbobaggins, 2018-06-15:
I get a communication error with the touch portion of the screen connected. The error is intermittent every few seconds it is present then will go to no error then error. Is there a way to fix? Please advise.

shawaj, 2018-06-15:
The touch should work just through the ribbon cable. Are you trying to connect it a different way?

tvoverbeek, 2018-06-15:
@opbillbobaggins Which Raspberry Pi model?
Every model with the 40-pin GPIO connector routes the I2C for the touch screen via the ribbon cable
and uses i2c bus 0 for the touch part of the display. So for those models i2c bus 1 (the one used by the PiJuice) should not be affected.
However If you use an older model (with 26-pin GPIO connector) you are sharing the I2C bus with the PiJuice since you have to connect the display i2c with jumper cables.
In case you have connected jumper wires for I2c on a 40-pin RPi model, remove them. Otherwise you certainly will get communication errors.

opbillbobaggins, 2018-06-16:
Thank you for the help. I had used the 4 wires supplied with the screen hooking them to the GPIO. Removing the 2 wires I thought were needed for the touch portion of  the screen fixed the issue. The touch screen still functions through the ribbon cable as you pointed out. Thank you again for the help in getting this resolved.

closed 2018-06-16 by opbillbobaggins