PiSupply/PiJuice #158: run in on  Pi3 b+

blanchloup22, 2018-06-15:
the run pin has been moves down due to the location of the new PoE added. The new location of the run pin is directly under the TC2030-IDC 6-Pin on the PJuice. I know this is not an issue that can be fixed on existing boards with out a small soldering of wires to the 2 places. Just a heads up in case it has not been noted yet.

shawaj, 2018-06-16:
Thanks! We've actually already redesigned the board and battery surround to accommodate this change.

However we already had a bunch of boards in production so it won't be bring introduced until a future production run.

Thanks for posting this though

closed 2018-06-16 by shawaj

blanchloup22, 2018-06-16:
Awesome! I figured this was actually the case but I wanted to say something
because I didn't see anything one it right off