PiSupply/PiJuice #159: Complete Power Off

karlsiegert, 2018-06-24:
I am looking to set a System Event on a low charge where the pijuice completely powers off the Pi...  where the only way to restart it is by plugging the micro USB power supply into the Pi  (just like you would if there were no UPS HAT).  Right now, all available System Event options that HALT/power off don't really shut the Pi completely down and thereby turning off the Pi's red Power LED light and the green LED light.  Those two LED lights seems to always be on even after a shut down.  And I need them off.

The best approximation of what I need is the HARD_FUNC_POWER_OFF script which is set by default to when someone presses the SW1 button for 20 seconds.  Is there any way to assign this same script to a low power System Event?

tvoverbeek, 2018-06-25:
Use a user function activated by low charge. and/or low battery voltage
In the user function do something like:
```
from pijuice import PiJuice
import os
pijuice = pijuice.PiJuice(1, 0x14)

# Remove power to PiJuice MCU IO pins
pijuice.power.SetSystemPowerSwitch(0)
# Remove 5V power to RPi after 60 seconds
pijuice.power.SetPowerOff(60)
# Shut down the RPi
os.system("sudo halt")
```
You can reduce the 60 seconds to e.g. 20 seconds, but it should be long enough
so the 'sudo halt' has sufficient time to bring the system down before the 5V power
disappears.

If you assign SYS_FUNC_HALT_POW_OFF to LONG_PRESS1 on SW1 you can try it out.
This shuts down the Pi and removes the 5V after 60 seconds.
Both green and red LEDs on the PI should be off then.

tvoverbeek, 2018-06-26:
Correction to my reply of yesterday:
In the PiJuice settings app (pijuice_gui.py or pijuice_cli.py) in the System Events tab
for the low charge event just select SYS_FUNC_HALT_POW_OFF.
This will shut down the RPi at low charge and after a minute remove the power to the RPi 5V.
When no power is connected to the PiJuice and the RPi (difficult to get a low charge event otherwise) the PiJuice will then enter a low power state indicated by a very slow blinking weak led.
If you want a shorter time before 5V power removal use the user function as indicated in my comment from yesterday.

closed 2019-09-15 by ChristopherRush