PiSupply/PiJuice #160: Wake up on charge question

richardquint, 2018-06-25:
In the Hardware/Readme.md I read

Wakeup on charge: When power source appears and battery starts charging this function can be configured to wakeup system when charge level reaches settable trigger level. Trigger level can be set in range 0%-100%. Setting trigger level to 0 means that system will wake-up as soon as power source appears and battery starts charging.

When I checked both System task enabled and Wake up on charge in the System Task dialog with Trigger level set to 0, then clicked Apply, waited a bit and then rebooted followed by a complete shutdown of the Pi and PiJuice I expected a power on when I applied power through the micro USB input on the PiJuice, but nothing happened.  I could use the button on the PiJuice to start the Pi up.  

I'll experiment with different trigger levels. but I wonder if I'm missing something.

ChristopherRush, 2018-06-28:
First of all make sure you are running the latest software v1.3.3. How did you shutdown the Raspberry Pi? A good test would be to use the "No Power" option under system events to do a SYS_FUNC_HALT_POW_OFF so that when you remove the power supply from the PiJuice the system will shutdown. You then need to wait a further 60 seconds for the PiJuice to completely remove power from the Raspberry Pi (Red LED will go off on the Pi). Then apply the power to the PiJuice and it should boot up based on "Wakeup on charge" 0%.

richardquint, 2018-06-28:
I'm attaching some screen shots that show my configuration and the curent state of software and firmware for my PiJuice hat.  The shutdown script (based on TVOVERBEEK's suggestion in response to issue #147) should be equivalent to your recommendation.

When I turn on power to the PiJuice hat LED1 starts flashing indicating that the battery is charging but to start the Pi I have to use a SINGLE_PRESS as configured for SW3.

BTW, if this should be a different issue I'll open it separately, but the Button dialog on a small screen doesn't inclde a scroll bar to get to all the options for SW3 or the Apply button.  Even if you move the window the bottom part of the dialog doesn't exist if you have a really small screen (I'm using an Adafruit 3.5" PiTFT). 
[image]
[image]
[image]
[image]
[image]
[image]
[image]

richardquint, 2018-07-03:
I have tried ChristopherRush's suggestion and turning off power after checking the "No  Power" option with the associated event SYS_FUNC_HALT_POW_OFF does nothing.  On the other hand, setting Minimum Charge to 4% and the low charge event to be my ToTalShutdown script works as it should, even though Wake on Charge still doesn't work. 

Should I try resetting the PiJuice to its initial state and resetting everything from scratch?  If yes, what is the best way to do it; just pull the battery, do the SW2 + SW3 simultaneous push, or both and in what order?

tvoverbeek, 2018-07-03:
Just tested Cristopher Rush's suggestion. Short: it works for me:

Started with power connected to the PiJuice USB input.
Setting 'No Power' in System Events to SYS_FUNC_HALT_POWER_OFF
Setting 'Wakeup on charge' enabled in System Task with trigger level 0%.
(System Task should also be enabled)
When I remove the power from the PiJuice USB input it takes a few seconds before LED D2 starts blinking red (=sending the 'sudo halt' command to the Pi), the Pi shuts down and about a minute later the red LED on the Pi goes off (=Power off for the Pi)
Then the Pijuice goes to low power mode: slow low intensity blink of LED D1.
When I then reconnect the power to the PiJuice USB input, the Pi boots immediately.

@richardquint Pulling and reinserting the battery restarts the firmware but settings are not changed.
Better to do the SW2+SW3 simulatenous push, which will restore default firmware settings and also restarting the firmware.

richardquint, 2018-07-08:
I've tried but I am confused about the order I should perform the different tasks to reset everything. What I mean is what should the state of the battery be, should the battery be connected or removed when I do the SW1+SW2 press, should there be power connected to the Pi or the PiJuice when I do the presses?  I would appreciate a step by step list.  Looking at other posts I suspect this could reduce a lot of the traffic.

richardquint, 2018-07-09:
Continuing the saga.  I have done the SW1+SW2 reset.  If I enable the system event "no power" to be SYS_FUNC_HALT_POWER_OFF then everything works as expected. Removing power to the PiJuice leads to a full shutdown.  If I wait a minute and then turn on the power the Pi boots as expected.  However if I uncheck the system event for no power I am back to the situation that I can't start the Pi by applying power to the PiJuice.  I can start the Pi by using SW3 (I set a press to be HARD_FUNC_POWER_ON) or by directly applying power to the Pi's microUSB input.

cmsimike, 2018-07-24:
Sorry if this isn't the 100% best place for this (I feel like it is vaguely connected), but is there a way to power down the raspberry pi manually (meaning I personally run sudo shutdown -P now), disconnect the power, and have the pijuice boot up the pi when the power is reconnected?

So far my tests don't seem like that is possible but I'm hoping I am missing something.

richardquint, 2018-07-28:
@tvoverbeek's comment for issues #176 and #176 have put me right.  However the documentation could be a clearer.  Just changing the settings in the GUI isn't sufficient to get the expected behaviour.

closed 2018-07-28 by richardquint

shawaj, 2018-07-29:
@ChistopherRush

Danielcor17, 2018-11-22:
This is not much of a UPS if you have to set the NO POWER system event to SYS_FUNC_HALT_POWER_OFF in order to get it to wake up on charging? This means that your pi will always shutdown in the event of no power. How can you make this UPS wake on charge after a software shutdown and not affect its ability to stay on during intermittent NO POWER events?