PiSupply/PiJuice #162: PiJuice External Battery set up

reddo10, 2018-07-04:
Hello,

I'm struggling with the set up of a single cell 10000mAh external battery to the PiJuice. Although I'm assured by the manufacturers they this will work, it seems to behave erratically. Has anyone successfully set up a higher mAh external battery? Could anyone kindly walk me through what they did?

Kind regards,

Justin

ChristopherRush, 2018-07-04:
@reddo10 What do you mean by "erratically" ?

reddo10, 2018-07-04:
If I set the battery up on the screw connector and plug in a mains power source, all seems fine. The Pi operates and the PiJuice shows all the right lights (blue charging light with intermittent green light showing battery is over 50% charged). When I pull the mains power away from the PiJuice the battery seamlessly kicks in and the Pi operation isn't interrupted. However, when I put the mains power back in the PiJuice this causes a reset to the Pi. Power seems to be dropped for a second or so. 

I've tried altering the DIP switch to Debbie the battery is lipo but to no avail. The battery is a 10000mAh lipo single cell battery.

I've not made any alterations or used the software. I only need a solar panel to charge the larger battery to guarantee Pi uptime. 

Any thoughts?

Any ideas?

reddo10, 2018-07-04:
*denote not Debbie....

closed 2019-09-15 by ChristopherRush