PiSupply/PiJuice #163: Reset through CLI causes crash

github-0, 2018-07-04:
My PiJuice is not functioning correctly (buttons no longer respond to anything) so I thought I'd try resetting. Documentation says
"Dual long press of SW1 and SW2 for 20 seconds will reset PiJuice HAT configuration to default. This applies to the MCU configuration only."

Just to confirm, "dual long press" means to press and hold both buttons? This had the same response as other functions, ie. nothing happened. Next I tried to reset through pijuice_cli.py. After choosing and confirming reset I got this:

`Traceback (most recent call last):
  File "/usr/bin/pijuice_cli.py", line 1320, in <module>
    loop.run()
  File "/usr/lib/python2.7/dist-packages/urwid/main_loop.py", line 278, in run
    self._run()
  File "/usr/lib/python2.7/dist-packages/urwid/main_loop.py", line 376, in _run
    self.event_loop.run()
  File "/usr/lib/python2.7/dist-packages/urwid/main_loop.py", line 682, in run
    self._loop()
  File "/usr/lib/python2.7/dist-packages/urwid/main_loop.py", line 715, in _loop
    alarm_callback()
  File "/usr/lib/python2.7/dist-packages/urwid/main_loop.py", line 164, in cb
    callback(self, user_data)
  File "/usr/bin/pijuice_cli.py", line 111, in update_status
    status_args = self.get_status()
  File "/usr/bin/pijuice_cli.py", line 83, in get_status
    gpio_info = "%.3fV, %.3fA, %s" % (io_voltage / 1000, io_current / 1000, status['powerInput5vIo'])
KeyError: 'powerInput5vIo'
`
It looks like the reset was succesfull though.

ChristopherRush, 2018-07-04:
@github-0 Try pulling the battery out to restart the firmware then reset the configuration by holding down both SW1 and SW2 for 20+ seconds or via the software.

tvoverbeek, 2018-07-05:
@github-0 Thanks for the Traceback. It is a bug in pijuice_cli.py. Looking for a fix right now.

ChristopherRush, 2019-09-15:
Fixed - 85bb941171d89c280fdee85e321a99bada6ec8ca

closed 2019-09-15 by ChristopherRush