PiSupply/PiJuice #164: pijuice bricked after successful firmware update with pijuiceboot

gepatto, 2018-07-05:
I tried to update my pijuice with the latest firmware using pijuiceboot:
`pijuiceboot 14 PiJuice-V1.2_2018_05_02.elf.binary`
There were no errors and pijuiceboot reported success.

shawaj, 2018-07-05:
There is a firmware reset process. Did you try that? Just hold SW1 and SW2 for 20 seconds

shawaj, 2018-07-05:
See here - https://github.com/PiSupply/PiJuice/blob/f66f7110ce0e89ef75b7de42ba0c60404c3e11ad/Hardware/README.md#special-functions

gepatto, 2018-07-06:
Yes I tried 
- holding sw1+sw2 for 20 seconds  -> nothing happens
- holding sw3 while connecting power to the pijuice -> nothing happens
- holding sw3 while connecting power to the pi -> nothing happens
- connecting to a different pi3 -> nothing happens
- removing the battery for 10 minutes and placing it again ->nothing happens

by nothing happens I mean nothing at all, not even a short flash of an led.
And it all starting right after the firmware upload was completed without errors 

side note: I still read 3.4volts over the screwterminals GND and VBAT 
( don't know if that helps debugging what might be wrong )

tvoverbeek, 2018-07-06:
@gepatto holding sw3 while connecting power to the pijuice should start the boot loader on the PiJuice. However you still need to send the firmware file using the pijuiceboot program.
Before you start pijuiceboot make sure the pijuice service is not running and also kill the pijuice_tray program when using the desktop. This to avoid interference from i2c commands from the service and the tray app when running pijuiceboot.

gepatto, 2018-07-06:
I'm starting to think the firmware somehow triggered a hardware defect. (don't now how)

@tvoverbeek: 
I stopped the pijuice service (not running a desktop) 
`sudo service pijuice stop`
held down sw3 while also connecting power to the pijuice and released the button after 3 seconds
but nothing happens, no leds and it's not showing up on i2cdetect
So when I run 
`pijjuiceboot 14 PiJuice-V1.2_2018_03_28.elf.binary`
it fails with
```
using address 14 
Input file PiJuice-V1.2_2018_03_28.elf.binary  
Starting bootloader  
Starting bootloader old firmware  
Sending get command  
error receiving data -1 
```

Just a thought: Could the battery somehow be the cause?
That's why I started the firmware update in the first place, because I was getting very fluctuating battery levels.

shawaj, 2018-07-06:
Just an FYI, the fluctuations in battery levels is a known issue and is due to the in built protection circuitry in the battery. We are looking into ways to improve this but it shouldn't cause any particular issues.

With the see and sw1/see tricks, did you make sure to remove the battery entirely, plug in to the PiJuice power and then perform those tricks? The bootloader sequence only happens after total power down and then a power on which won't happen if the battery is in.

tvoverbeek, 2018-07-06:
Just a note: To stop the service use the command
`sudo systemctl stop pijuice.service`
You can check the status with:
`systemctl status pijuice.service`

gepatto, 2018-07-06:
Ok, so I

- stopped the pijuice service: `sudo systemctl stop pijuice.service`
- removed the battery
- disconnected all power
- left the pijuice for 15 minutes without battery and power and disconnected from the pi

now I'm about to
- connect the pijuice to the pi
- hold down sw3
- connect power the pijuice (not the pi)
- wait 10 seconds before releasing sw3  (or should I wait longer or shorter)
- **should I see some feedback?**
- let the pi boot
- make sure pijuice service isn't running:  `systemctl status pijuice.service`
- **should i run `i2cdetect -y 1` to discover the i2c port number or use default 14?**
- run `pijjuiceboot 14 PiJuice-V1.2_2018_03_28.elf.binary`

shawaj, 2018-07-06:
@ChristopherRush I'm wondering whether we should document some of this firmware update procedure more heavily in order to have this all in one place. As it's a bit piecemeal ATM. Maybe under the firmware readme? Might be duplication of some bits but that's not the end of the world.

shawaj, 2018-07-06:
@gepatto the last command has a typo - only one j in pijuiceboot - other than that it looks good to me, unless @tvoverbeek can spot any other things

shawaj, 2018-07-06:
Oh and default 14 should work without needing to detect yes.

gepatto, 2018-07-06:
yes the 2 j's was a typo while making the post.

Ok, fingers crossed here I go..

gepatto, 2018-07-06:
```
EEPRM programming finished successfully
Code executed successfully
```
Do I need to do something else?
or just:
shutdown,
reconnect the battery
power on using power connected to the pijuice

gepatto, 2018-07-06:
by the way. 
I held sw3 for 10 seconds.. not sure if there a specific amount of time I needed to hold it

shawaj, 2018-07-06:
Yes that should be fine. Maybe remove the power completely after the shutdown

gepatto, 2018-07-06:
No Success
Now the Pi doesn't even boot .
Battery inserted, power connected to the pijuice.

I'm disconnecting it again, removing the battery, leave it unpowered for 10 minutes and start over.

gepatto, 2018-07-06:
I repeated the process, but again with no success.
Here's the output of pijuiceboot 
[pijuiceboot.log](https://github.com/PiSupply/PiJuice/files/2170381/pijuiceboot.log)

I have sent an email to sales@pi-supply.com and received this ticket #44952

Thanks for all your help so far!
If there's more I can do to help prevent other people from having the same kind of trouble 
please let me know.

shawaj, 2018-07-06:
My colleague David will arrange to get a replacement sorted and to get the old one back at our cost for further debugging. Will leave this issue open until we work out if there was a reason the firmware update caused an issue

gepatto, 2018-07-10:
@shawaj I received my replacement today :) and the bricked one is on it's way to Unit 4.
I see the firmware on the new one is v1.1 but I think I'll leave it at that until we now what went wrong.
Thanks for all the help !

_Here's the story about how I think I got my PiJuice Bricked:_
I noticed that battery level was fluctuating with steps of at least 10%.
Checking the firmware version I saw that was at version 1.1
At that time I was running a desktop so I upgraded the firmware to version 1.2 
through the PiJuice-GUI dialog.
After updating the firmware I noticed that the batterylevel still fluctuated and that
,strangely enough, the firmware-tab from the PiJuice-GUI dialog had disappeared!
I had to leave it at that for that moment as I had to go to work (on location).
I brought the Pi and Pijuice with me and decided to retry uploading the firmware using pijuiceboot.
(because I didn't bring a display and didn't think of using vnc)
I downloaded PiJuice-V1.2_2018_05_02.elf.binary, stopped the pijuiceservice and used  
`pijuiceboot 14 PiJuice-V1.2_2018_05_02.elf.binary`
And that was the end of my PiJuice. All leds turned of and from there on it was bricked and not showing up in i2cdetect anymore.

This is where this thread started.

ChristopherRush, 2018-07-10:
@shawaj Just to confirm do we need to enter boot loader mode every time we flash the firmware or just when there are issues with the normal process?

shawaj, 2018-07-10:
Just when there are issues. It should work from the GUI or the command line usually

ChristopherRush, 2018-08-03:
40a2eb56787afbb723933a78ad3544cb3615ba96

closed 2018-08-03 by ChristopherRush