PiSupply/PiJuice #166: need help guidance on how to change default I2C address of RTC

hemingchen, 2018-07-09:
I was trying to use PiJuice onboard RTC and another HAT that has a MPU-9050 chip. Unfortunately both PiJuice and the MPU-9050 run on I2C 0x68 by default causing a conflict. While I'm having trouble moving MPU-9050 to a different address, could someone guide me on how to change the PiJuice RTC I2C address - say 0x67 or something? 
I tried changing the RTC address in PiJuice GUI but somehow the OS still thinks the RTC is at 0x68 and wouldn't work with the new address. Do I need to change the EEPROM address as well so the OS wouldn't load the default HAT config? 
Thanks!
-Heming

shawaj, 2018-07-09:
You can do that from the PiJuice HAT General Config Menu - https://github.com/PiSupply/PiJuice/tree/master/Software#pijuice-hat-general-config-menu

closed 2018-07-09 by shawaj

tvoverbeek, 2018-07-09:
@hemingchen Changing the RTC address from 0x69 to e.g. 0x68 in the GUI works.
However if you then do `i2cdetect -y 1` you will see a 'UU' entry for 0x68 and a new entry for 0x69.
The 'UU' comes from the RTC kernel module which was loaded at boot due to the device-tree fragment in the PiJuice id-eeprom (when you leave the PiJuice id-eeprom address at 0x50).
When you reboot the 'UU' entry will be gone.
You will have to change the id-eeprom address to 0x52 anyway because otherwise there will be a conflict between the two id-eeproms from the PiJuice and your MPU-9050 HAT.

On a side note: Changing the PiJuice I2C address and the PiJuice RTC I2C address does not work in pijuice_cli (the change setting code is disabled in pijuice_cli). Will open a new issue for this.

shawaj, 2018-07-09:
@tvoverbeek thanks! Maybe we should add the eeprom device-tree fragment to the official kernel so you can load it from boot/config.txt as well, with an override for the RTC address (and perhaps have this in the software / firmware too....so it does not auto-load the RTC on an address it is not using?

tvoverbeek, 2018-07-09:
@shawaj The PiJuice id-eeprom device-tree fragment specifies address 0x68.
There is already  device-tree overlay for the ds1339 (`i2c-rtc,ds1339=1`) which also has a fixed i2c address of 0x68 (as the real device does).
When booting the RPi with a changed RTC I2C address the RTC init for the ds1339 will fail (with the PiJuice id-eeprom at 0x50).

Adding an overlay where you can set the I2C address would be nice, so you can still use the linux hwclock support.

shawaj, 2018-07-09:
so it sounds like we would need to update the ds1339 i2c address to being updatable, as well as the pijuice one?

shawaj, 2018-07-09:
@tvoverbeek discussing this at above link with RasPi team

hemingchen, 2018-07-09:
@tvoverbeek Thanks very much! What you described in your first comment above was exactly what I experienced. I eventually changed PiJuice to 0x52 and RTC to 0x67 (both in PiJuice GUI) and removed `dtoverlay=i2c-rtc,ds1339=1` from config, then hwclock works after doing a `echo ds1307 0x67 > /sys/bus/i2c/devices/i2c-1//new_device`. I had to add this line to `/etc/rc.local`, ugly but so far works ok for me.
Thanks to everyone for the support.

tvoverbeek, 2018-07-12:
@hemingchen @shawaj
@pelwell (Phil Ellwell) at the Raspberry Foundation has added the option to set the address in the i2c-rtc device-tree overlay.
This is now available if you do a `sudo rpi-update`.
With this overlay you can specify in /boot/config.txt  e.g. `dtoverlay=i2c-rtc,d1339=1,addr=0x67`
If you do not want to run `sudo rpi-update` you will have to wait for the next update of the raspberrypi-kernel package or use the @hemingchen work-around above..

pelwell, 2018-07-12:
You can also download just the .dtbo file [here](https://github.com/Hexxeh/rpi-firmware/blob/master/overlays/i2c-rtc.dtbo) - copy it into the `overlays` directory, add the `addr` parameter in `config.txt`,  and reboot.

hemingchen, 2018-07-12:
@pelwell @tvoverbeek @shawaj thank very much for the quick update. It now works - after installing the latest `.dtbo` file to `/boot/overlays/` and adding `dtoverlay=i2c-rtc,ds1339=1,addr=0x67` to `/boot/configure.txt`.

JJSlabbert, 2019-03-26:
You can also create a second i2c bus from free gpio pins
https://github.com/JJSlabbert/Raspberry_PI_i2C_conficts
https://www.instructables.com/id/Raspberry-PI-Multiple-I2c-Devices/