PiSupply/PiJuice #168: Setting SW1, SW2 or SW3 to work with user scripts...

rossirm, 2018-07-10:
Hi to all!
I am Rossi from Bulgaria and began to use PiJuice 2 weeks ago.
I have a system of Raspberry Pi 3B model + PiJuice + Sense Hat. All works well instead of the fact that I can't say to button SW3 for example to execute a python script which is in the directory /home/pi
I show the path to the PiJuice software/User Scripts like /home/pi/python3 file_name.py
But the result is that when I press button SW3 it doen's work although I used commands in the terminal chmod +x file_name.py and sudo 777 /home/pi/file_name.py
PLEASE HELP ME! I want it to work very much.
Perhaps this is not so big problem for you, but I am new to Raspberry world and sorry for my English if it is not so good!

closed 2018-07-10 by rossirm

shawaj, 2018-07-10:
@rossirm did you fix this?

rossirm, 2018-07-10:
Yes, it works. I have written #!/usr/bin/python3 instead of only python and for the path in user scripts functions I have written only python file_name.py instead of the full path of the file and the file_name without python before the its title.

shawaj, 2018-07-10:
Great, thanks for the update!