PiSupply/PiJuice #169: System Time- Unable to set it from pijuice_cli.py

dducret, 2018-07-10:
When I select < Set System time >  nothing ....
Is there a workaround to set the time on PiJuice ?

tvoverbeek, 2018-07-10:
@dducret < Set System time > is misleading. The command copies the Raspbian time to the hardware clock (RTC) on the PiJuice.
The time displayed on the tab is the hardware clock time in UTC.
If the time is already in sync, you indeed see nothing change.

shawaj, 2018-07-10:
Maybe we should change it to say copy system time instead?

dducret, 2018-07-10:
Date on RPI Tue 10 Jul 14:57:54 CEST 2018
On cli: UTC Time: Tue 2018-07-10 12:58:19

tvoverbeek, 2018-07-10:
@shawaj @dducret 
Already in github. since April 14.
Now it says 'Set RTC time'
See commit https://github.com/PiSupply/PiJuice/commit/7f482443778e48b4f9caeb19745f32c67ed83abf from April 14

tvoverbeek, 2018-07-10:
@dducret Obviously you do not have the latest version of the pijuice-gui package.
The 'Set RTC time' change is already in version 1.3.3 of the package.

closed 2019-09-15 by ChristopherRush