PiSupply/PiJuice #170: ImportError: cannot import name PiJuice

ChristopherRush, 2018-07-10:
Note sure as to why I'm receiving this error on a fresh install of Stretch 2018-04-18. PiJuice has firmware v1.2 and has configuration settings reset. No other HATs are installed and ran `apt get update` then `apt get install pijuice-gui` . 

Python script is:

`from pijuice import PiJuice`
`pijuice = PiJuice(1, 0x14)`
`print pijuice.status.GetStatus()`
`print pijuice.status.GetChargeLevel()`

tvoverbeek, 2018-07-10:
You can get his error if you have  a circular dependency.
However I do not see how you can get this with your script.
What other files are in the directory where your script is located?
An other pijuice.py ???

tvoverbeek, 2018-07-10:
I get the error if I give your script the filename pijuice.py

ChristopherRush, 2018-07-10:
@tvoverbeek Yes, so it would seem. You cannot name your scripts pijuice.py

closed 2018-07-10 by ChristopherRush